MarlinFirmware/Marlin #19272: [BUG] M114 (and other realtime commands) are hard and ambiguous to detect & parse

duk3luk3, 2020-09-06:
### Bug Description

I am coming from this: https://github.com/kliment/Printrun/issues/1103

Problem:
- Host programs have to parse realtime / monitoring output from Marlin
- There is no convention/protocol to make output from Marlin explicitly "machine-readable"
- To detect M114 output, Host programs (at least pronterface) search for "Count" in Marlin output
- This means any other Marlin output containing the string "Count" potentially leads to problems with output being suppressed or - worse - host programs misparsing output

Would it be viable to introduce a new convention to explicitly mark command output in a way that removes ambiguity.

For example, commands could put their command name at the start of the output, so M114 output would instead of:

```
X:110.00 Y:110.00 Z:8.00 E:0.00 Count X:8800 Y:8800 Z:12800
```

Print:

```
M114: X:110.00 Y:110.00 Z:8.00 E:0.00 Count X:8800 Y:8800 Z:12800
```

### My Configurations

N/A

### Steps to Reproduce

1. Flash any probe-requipped printer with probing and G35 firmware (`#define ASSISTED_TRAMMING`)
2. Get pronterface at [ff056a0](https://github.com/kliment/Printrun/tree/ff056a06dbdd9cd5ee7ab5f3cccccb5025d715fe) or earlier
3. Run G35

**Expected behavior:**

G35 output with "Clockwise" and "Counter-clockwise" adjustment instructions:

```
Probing point 0 (Front-Left) X19.00 Y21.00 Z-0.05375
echo:busy: processing
Probing point 1 (Front-Right) X190.00 Y21.00 Z-0.24500
Probing point 2 (Back-Right) X190.00 Y195.00 Z-0.25625
echo:busy: processing
Probing point 3 (Back-Left) X19.00 Y195.00 Z-0.23375
Evaluating Front-Right
Evaluated Front-Right: diff = 0.19125; adjust = 0.27321
Turn Front-Right
 Clockwise
 by 0 turns
 and 16 minutes (98 degrees)
Done Front-Right
Evaluating Back-Right
Evaluated Back-Right: diff = 0.20250; adjust = 0.28929
Turn Back-Right
 Clockwise
 by 0 turns
 and 17 minutes (104 degrees)
Done Back-Right
Evaluating Back-Left
Evaluated Back-Left: diff = -0.00688; adjust = -0.01375
Turn Back-Left 
 Counter-Clockwise               
 by 0 turns    
 and 0 minutes (4 degrees)
Done Back-Left 

```

**Actual behavior:**

All lines containing "Counter-Clockwise" are suppressed:

```
>>> G35 V2 Z0 S40
SENDING:G35 V2 Z0 S40
Probing point 0 (Front-Left) X19.00 Y21.00 Z-0.23062
Probing point 1 (Front-Right) X190.00 Y21.00 Z-0.22187
Probing point 2 (Back-Right) X190.00 Y195.00 Z-0.24687
Probing point 3 (Back-Left) X19.00 Y195.00 Z-0.22937
Evaluating Front-Right
Evaluated Front-Right: diff = -0.00875; adjust = -0.01250
Turn Front-Right
 by 0 turns
 and 0 minutes (4 degrees)
Done Front-Right
Evaluating Back-Right
Evaluated Back-Right: diff = 0.01625; adjust = 0.02321
Turn Back-Right
 Clockwise
 by 0 turns
 and 1 minutes (8 degrees)
Done Back-Right
Evaluating Back-Left
Evaluated Back-Left: diff = -0.00125; adjust = -0.00179
Turn Back-Left
 by 0 turns
Done Back-Left
```

mron, 2020-09-06:
M114 is an immediate command?

boelle, 2020-09-09:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

closed 2020-10-23 by github-actions[bot]