MarlinFirmware/Marlin #19273: [BUG] USB communication with Linux host is randomly interrupted

bkiziuk, 2020-09-06:
Hallo,
I recently started to use 3D printer with MKS Robin Nano 1.2 board.
The Marlin version is Release 2.0.6.1.
I was trying to print from Repetier-Host application under Linux Mint 20 x64. The version of Repetier-Host is 2.1.6 built as AppImage.

### Bug Description

Repetier-Host randomly loses communication with the printer. The printer stops the print. Repetier-Host no longer gets status updates from the printer.
It is needed to disconnect and reconnect via the Repetier-Host Disconnect/Connect button to regain communication.
USB trace on the Linux host reveals that as soon as URB_INTERRUPT appears on the bus, the printer stops to send status updates.

### My Configurations

Configurations and debug traces included.
[MKSRobinNanoUsb.zip](https://github.com/MarlinFirmware/Marlin/files/5179634/MKSRobinNanoUsb.zip)

MKS Robin Nano is visible as:
Bus 005 Device 003: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter

### Steps to Reproduce

In my environment it is enough to leave the printer idle for some time connected to Repetier-Host and make it just report the status. The time range varies from minutes to hours.

boelle, 2020-09-09:
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

sjasonsmith, 2020-09-11:
@bkiziuk, a change went into `bugfix-2.0.x` a few days ago which fixed hangs on STM32F1 boards like yours. They were specifically related to serial communication.

Please re-test as @boelle requested. In this case we know that a related fix was recently made, and will presume this is resolved if we do not hear back from you.

This is the change:
#19301

bkiziuk, 2020-09-11:
I'm testing it right now. No conclusions yet as it takes some time. I'll keep you posted.

bkiziuk, 2020-09-12:
Unfortunately, the situation did not improve after moving to the yesterday's bugfix-2.0.x.
New files attached. I'm not really sure how to get more info on it :-(
[bugfix2-0-x_usb_20200912.zip](https://github.com/MarlinFirmware/Marlin/files/5212325/bugfix2-0-x_usb_20200912.zip)

Squeeze, 2020-10-14:
Is there any news on this?
It looks like I'm having exactly the same issue with my Fysetc Cheetah board in a Creality Ender 3 Pro.
Whatever device I tried, it looses its connection randomly. I've used Octoprint (Octopi on a Raspberry Pi 3b) and it worked for over a year without any problems. I'm not 100% sure but I believe the last version it works without connection problems was 2.0.4.1. But in that version I couldn't use the sd card.
I've tried to connect via Repitier Host and Simplify3D and even tried a new Raspberry Pi 4B with a new instance of Octopi, but no success.
On this moment the board has Marlin 2.0.7 on it, and I also tried bugfix 2.0.x from about a week ago. All with the same result.

bkiziuk, 2020-10-14:
In the mean time I made some hardware changes (replaced 2xTMC2208+2xA4988 with 4xTMC2209 in hardware UART mode), upgraded again to the latest bugfix-2.0.x at the time and replaced Repetier software with Printrun/Pronterface.
After these changes I'm still observing the issue (the Linux host PC is the same as before).
I'll edit this comment with new files I caught a few days ago.

Clarenceguoshuai, 2020-11-15:
但呼吁依旧

rhapsodyv, 2020-11-18:
Do you have another OS or board to test? Just to try to exclude OS and hardware as main problem.

boelle, 2020-11-18:
> 但呼吁依旧

English works better

bkiziuk, 2020-11-19:
@rhapsodyv I'm trying to check it. It will take some time.

bkiziuk, 2020-11-20:
I don't have access to another RobinNano board, however I also checked different PC, also with different OS.
I can see the same issue - at least same effect as USB traces are bit different. It is beyond my abilities to correctly interpret the correctness of USB communication.

Tested configurations:
1) original configuration
OS: Linux Mint 20 x64
Hardware: Desktop Dell PC
Capture: usbmon + libpcap/tcpdump
(data attached before)
2) alternative Linux
OS: Linux Mint 20 x64
Hardware: Dell Latitude 5420
Capture: usbmon + libpcap/tcpdump
3) Windows machine
OS: Windows 7 x64
Hardware: Fujitsu H730
Capture: usbpcap 1.5.4.0

[config.zip](https://github.com/MarlinFirmware/Marlin/files/5572021/config.zip)

Maybe someone can contribute to this issue with data from another RobinNano board.

meteozond, 2020-12-02:
Got this with Debian+OctoPrint and MacOS+RepetierHost on latest 2.0.x bugfix.
There is no any system. It could work for days and lost connection after 15 min since print start.

sjasonsmith, 2020-12-04:
The USB port on these boards are not controlled by Marlin at all. Nothing in Marlin can control whether the connection between the host and board is functioning properly. The USB controllers are standalone chips which do nothing but provide UART RX/TX lines to the MCU Marlin runs on.

Your logs show a USBD_STATUS_XACT_ERROR. In my experience these are _extremely_ difficult to recover from. They happen due to low-level failures in the host controller. When I have encountered them in the past they occurred if another device (such as a USB Mass Storage Device) was inserted or removed during a transfer. Once they occur, the device driver and firmware are in an inconsistent state. In some cases the drive and firmware can recover. This board has an extremely low-cost USB-Serial converter, so I am not at all surprised that it does not handle recovery attempts well.

Some ideas to try to avoid this issue in the first place:
1. Make sure you are using high quality USB cables.
2. Try different USB ports. If you are using USB 3.0 ports, try a USB 2.0 port.
3. Try connecting through a USB hub.

Unfortunately there isn't anything we can do to make it better. If anything could be done it would have to be done in the device driver and in the actual USB-Serial converter chip, not in Marlin.

closed 2020-12-04 by sjasonsmith