MarlinFirmware/Marlin #19274: [BUG] (MKS Robin v2.3 board resets mid print)

mattdog01, 2020-09-06:
### Bug Description

During a print, usually early in the print (10+- 5 minutes in roughly) the printer will perform one or two random travels, off the print to one end of an axis and then back to the print (similar to a corrupted SD card line error) and continue printing.  Shorty (minute or so) after this, it stops without any further movement of the printhead or extruders. Then **does the "reset" where the board and screen turn off and then back on as though the power was just cycled off and then on**, or as though the on-board reset button was pushed.  The screen returns to its normal "just after power on" condition, "?" marks flashing in the position indication spots, heaters setting is back to "0" and the temps are dropping to room temperature.

### My Configurations

The board is the MKS Robin v2.3 with MKS TFT v1.1R 
coreXY printer
running on 24volts

files attached

### Steps to Reproduce

1) turn on the printer normally
2) run "SD" in custom commands to enable reading the SD card
3) tap on the SD card icon on the status page
3) browse to the desired g-code to print and start the print 
4) print starts normally prints the first few layers
5) it stops without user input
6) the board restarts on its own

Resets when...
Printing from an SD card in New Marlin GUI mode

Doesn't reset when...
Printing from Pronterface in New Marlin GUI mode
Printing from SD card in classic Marlin UI mode

Build enviro for New GUI is mks_robin_stm32
Build enviro for Classic UI is mks_robin

**Expected behavior:**
Complete the print without stopping.

**Actual behavior:**
Stops mid-print without raising.
If allowed to, the nozzle is stuck in the cooling print.

#### Additional Information
STL is from Fusion 360
G-code created by PrusaSlicer
No issues with hundreds of other prints from these sources.

Note: used the Discord #support channel to attempt a resolution to no avail.  Username Mattdog01.

[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/5179723/Configuration.files.zip)

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

mattdog01, 2020-09-06:
This the end of the log as the printer stops moving and goes into reset (there were literally thousands of Error:SD read error that are not included in this file) note that there is a Watchdog reset triggered by the fault.
[message.txt](https://github.com/MarlinFirmware/Marlin/files/5180392/message.txt)

This is the file that I was trying to print.
[Nozzle Fan - points down.zip](https://github.com/MarlinFirmware/Marlin/files/5180396/Nozzle.Fan.-.points.down.zip)

nanotuxi, 2020-09-07:
Similar issue with 8-bit MKS Gen L v1.0 with Marlin 2.0.6.1 from 2.0.x
Printing large files (about 60MB) from sd print stops in the middle of the print with
`SD init fail` shown on the LCD screen.
AFAIK examining the source code for the sd card part in Marlin2.x.x had some major changes during the last weeks, so I reverted back to an older  2.0.5 version and everything is o.k.
Maybe an issue with 2 sd cards option  on a 32-bit board (Onboard / LCD sd card) and having only a single (LCD sd card) on an 8.bit board?

mattdog01, 2020-09-07:
@nanotuxi those are some good thoughts.  I didn't even think about file size being an issue.  I will try some different size files as well.  The Robin v2.3 board only has one actual physical SD slot, on the main board, but I hear what you are saying.  The more ability a board has the more room for issues. 
It will be great if someone is able to find the issue and fix it and it is nice to hear that more than just my board has been affected as that will move it higher on the "lets look at this one" list.

as a side note
The community of developers are amazing and they don't get enough thanks.  I hope they don't get too discouraged by the complaints and understand that the Marlin user base is so appreciative of them.  I was working with the folks on discord and they were awesome helping me test and check.

AnHardt, 2020-09-07:
>  I didn't even think about file size being an issue.

It isn't.
Marlin makes use of the file-size at only one point - when reading a directory and telling the host. At no point Marlin is aware of how many bytes, or commands, may remain in the command-sequence, regardless of coming from file or host. Even all the kinds of progress indicators are only displayed, but not used otherwise.
Just, if there is a little chance there is a random read/communication error, the probability to fail a command-sequence rises with their length.

mattdog01, 2020-09-07:
> It isn't.

best phrase I've heard in a long while. 
With a large print volume and complex geometry those file sizes can get scary large.  But I get what your saying the more times you read from the card the more opportunities for it to not work.  
This issue it does not seem to be related to the media, an error in the g-code, the size of the code, or the actual SD slot (and its supporting circuits), as they all work in other print configs.  The only thing I can't rule out 100% is that if any of them are on the edge of being flakey then they may work in some situations and not is others. Clearly, I don't understand the coding end of the situation and that is where you fine folks come in.  Thanks for your help.

AnHardt, 2020-09-07:
> With a large print volume and complex geometry those file sizes can get scary large

Marlin does not know about print volume, complexity, size and does not care about that terms. It just and only sees the current command.

Imagine 2 files. One with a little simple part. The other with this part copied 4 times to the plate. Then the second file is about 4 times larger than the first. Then the chance of failing the print of the second part within the first quarter stays the same as that of failing the first part when printed completely.

Any chance of communication errors is bad.

nanotuxi, 2020-09-07:
Thanks for the information. But that doesn't solve the problem why the sd card error appears during middle of prints randomly. With smaller printsizes this problem never appears. 
So is it a problem with the hardware or is it software related? The card used has been reformated with only one single gcode file.

rhapsodyv, 2020-09-07:
I had a similar problem with chitu board. I don’t know you yours support it, but it solves for me:
```
#define SDIO_CLOCK                       4500000
#define SDIO_READ_RETRIES                     16
```

nanotuxi, 2020-09-07:
Maybe I should reduce the spi speed in marlin. My old 8 bit board seems to be flooded on the spi bus with too much information at a time. Maybe I should give it a try.

sjasonsmith, 2020-09-07:
I just realized there is now a `mks_robin_stm32` environment which uses the STM32 HAL. I happened to run across this change that was made when it was introduced:

```
  /*
    SDIO_INIT_CLK_DIV is 118
    SDIO clock frequency is 48MHz / (TRANSFER_CLOCK_DIV + 2)
    SDIO init clock frequency should not exceed 400KHz = 48MHz / (118 + 2)
    Default TRANSFER_CLOCK_DIV is 2 (118 / 40)
    Default SDIO clock frequency is 48MHz / (2 + 2) = 12 MHz
    This might be too fast for stable SDIO operations
    MKS Robin board seems to have stable SDIO with BusWide 1bit and ClockDiv 8 i.e. 4.8MHz SDIO clock frequency
    Additional testing is required as there are clearly some 4bit initialization problems
    Add -DTRANSFER_CLOCK_DIV=8 to build parameters to improve SDIO stability
  */

  #ifndef TRANSFER_CLOCK_DIV
    #define TRANSFER_CLOCK_DIV (uint8_t(SDIO_INIT_CLK_DIV) / 40)
  #endif
```

@mattdog01, which environment are you building with?

rhapsodyv, 2020-09-07:
Not only lowering speed, but increasing the max retry count too.

nanotuxi, 2020-09-07:
That seems to beva very good idea.
Thanks.

mattdog01, 2020-09-07:
@sjasonsmith Hey Jason,  I am building in the mks_robin_stm32 environment.  I understand that I need to use that eviro to get the super cool new Marlin GUI. :) I caught the printer in another loop this time it was in a g-code loop where it keeps performing a series of print head moves and then repeats.  I didn't get the serial output though, sorry.  But I did capture a short video that shows the print head stopping and then the printer resets.  I'll link to it once it is uploaded.

mattdog01, 2020-09-07:
I tried the SPI speed config (the half and quarter) Half didn't seem to help and quarter seemed to make it worse, AKA reset sooner.  

  * SD CARD: SPI SPEED
 *
 * Enable one of the following items for a slower SPI transfer speed.
 * This may be required to resolve "volume init" errors.
 */
//#define SPI_SPEED SPI_HALF_SPEED
//#define SPI_SPEED SPI_QUARTER_SPEED
//#define SPI_SPEED SPI_EIGHTH_SPEED

mattdog01, 2020-09-07:
@rhapsodyv Hey Victor, were you commenting on @nanotuxi 's situation or this error topic 

> I had a similar problem with chitu board. I don’t know you yours support it, but it solves for me:

Should I be trying your recommendation as well, or did I already with the SPI speed text I posted?

sjasonsmith, 2020-09-07:
> I am building in the mks_robin_stm32 environment.

OK, that puts you in an extremely small group of people then. It's entirely plausible there is something wrong with the SDIO configuration. @rhapsodyv do you know if it would be practical to just disable SDIO to see if it prevents the issue? I have no idea if this board would be configured properly to allow SPI to be used instead.

rhapsodyv, 2020-09-07:
We need to look at the board schematics to see if the spi pins for the SD aren't in use for others features.
But if the SD spi device + SD cs pin are available (and connected), we can surely disable SDIO and test SD SPI.

My chitu board use SDIO. And I had random errors, freezes and stops, while printing. After I lowered the SDIO clock and increased the retry count, my problems stopped.

mattdog01, 2020-09-07:
> OK, that puts you in an extremely small group of people then.
that sounds like i'm special but the reality......

I understood that SDIO needs to be enabled to get the 
#define SDIO_SUPPORT

@rhapsodyv do you need the board drawings?  Or do you already have them? MKS has those as open source now (I'm guessing you knew that though, LOL sorry).

@jmz52 has been working on the MKS-Robin implementation and with the new GUI here Support for TFT and touch screen - need feedback and testing #18129

sjasonsmith, 2020-09-07:
> OK, that puts you in an extremely small group of people then.
> that sounds like i'm special but the reality......

My point is just that the STM32 environment is only about two weeks old. You might only be the second person to use it :)

mattdog01, 2020-09-07:
JK JK  thanks @sjasonsmith that makes me feel better.  Bah Ha Ha

mattdog01, 2020-09-07:
link to the video of one of the times it has performed the Watchdog reset.  actual reset it at 2:15 

https://www.youtube.com/playlist?list=PLnEIv9L44zuAzhxCTtV6zslu_4Wr87Qre

jmz52, 2020-09-07:
@mattdog01, there are two possible causes for watchdog reset - UI buffer overrun and SDIO stability issue.
SDIO implementation for HAL STM32 is rather new not exactly stable, I had to force it to 1 bit mode to make is a bit more stable.
Try to add `-DTRANSFER_CLOCK_DIV=8` to your build parameters.

If that does not help, please compile `mks_robin_stm32` version  with no TFT screen and start print from SD card using pronterface.

mattdog01, 2020-09-07:
@jmz52 thanks Alexander.  I will give that a shot as soon as I'm able.  So let assume that I am an aircraft mechanic and not a coder, you know just in case there is an aircraft mechanic out there that is trying to do this, where would I put that line of code?  -DTRANSFER_CLOCK_DIV=8

Here in PlatformIO?  It looks like it is already there under build flags.

# MKS Robin (STM32F103ZET6)
# Uses HAL STM32 to support Marlin UI for TFT screen with optional touch panel
#
[env:mks_robin_stm32]
platform             = ${common_stm32.platform}
extends              = common_stm32
board                = genericSTM32F103ZE
board_build.core     = stm32
board_build.variant  = MARLIN_F103Zx
board_build.ldscript = ldscript.ld
board_build.offset   = 0x7000
board_build.firmware = Robin.bin
build_flags          = ${common_stm32.build_flags}
  -DENABLE_HWSERIAL3 -DTRANSFER_CLOCK_DIV=8
build_unflags        = ${common_stm32.build_unflags}
 -DUSBCON -DUSBD_USE_CDC
extra_scripts        = ${common.extra_scripts}
  pre:buildroot/share/PlatformIO/scripts/generic_create_variant.py
  buildroot/share/PlatformIO/scripts/stm32_bootloader.py
  buildroot/share/PlatformIO/scripts/mks_encrypt.py
lib_deps             =

mattdog01, 2020-09-07:
@jmz52 Disabled two features that required the display.  
Compiled in the mks_robin_stm32 enviro with TFT not selected, I didn't uncomment the touchscreen code but I won't touch the screen.  
Not sure if it needs to be uncommented as well or if it is ignored without the TFT.
Started a small print ( 1 hr ) to see but I may need to start a bigger print to get it to restart.  
I didn't want to commit to an all night print if it could have issues.  
If this works fine, then in the morning I'll start a longer print.

mattdog01, 2020-09-08:
@jmz52 the print finished just fine.  There were a couple odd codes from the serial data that are not in the g-code file but the file did complete and the print is fine.  I will try another longer print tomorrow using the same method.  
"MM27" happened twice

SD printing byte 1539162/7082468
SD printinSD printing byte 1543816/7082468
SD printing byte 1546260/7082468

SD printing byte 3400329/7082468
SD printing byte 3401449/7SD printing byte 3404212/7082468
SD printing byte 3405389/7082468

nanotuxi, 2020-09-08:
Which version of platformio is adviced to be used for compilation. 4 or 5?

mattdog01, 2020-09-08:
@jmz52 
[print log with TFT disabled in firmware.pdf](https://github.com/MarlinFirmware/Marlin/files/5189931/print.log.with.TFT.disabled.in.firmware.pdf)

Here are the results of another attempt to print with the TFT disabled in the firmware and starting an SD print.  
Note:
there a several times where MM27 comes up as an unrecognized command. MM27 does not seem to be in the G-code that I can find with a ctrl-F check of the file. 
There does seem to be some instability to the SD card reading as there were a couple axis runs of the printhead where the printhead is printing and the printhead run to the end of the axis either min or max. (X seems to happen more often but it does happen in Y and have never seen it in the Z)
The print was ended with the Watchdog reset and was not completed.

mattdog01, 2020-09-09:
@jmz52
I completed a print this time. It was a 12 hour print and the only issue was a layer shift caused by one of those random axis travels of the print head.  It was an X axis max travel but the print head hit something I had at the end of the X axis rails and this caused a X and Y layer shift (coreXY machine).  I promptly remove the obstruction on the end of the rails and the print continued  with only the one 2nd or 3rd layer shift.  Several times (3 to 5 times) it did a random axis travel to either a min end or max end on the X and Y axis.  This indicates, to me anyways, that there is still a slight issue with reading the SD card.  The card is fairly new  and had a full format completed before this print.  So i think that has been tested.  It is better without the TFT but still not perfect.  @jmz52, or anyone, what are the next steps you would like to take? @sjasonsmith anything you wanted to try?  I think you are working on the STM32 HAL right?

nanotuxi, 2020-09-09:
@mattdog01 
>... a 12 hour print and the only issue was a layer shift caused by one of those random axis travels of the print head.

>  print continued with only the one 2nd or 3rd layer shift.

This behaviour makes the software unusable at all, doesn't it? Nothing worth to mention if you print some sculptures, but for printing machine parts this is a no-go. So I hope it is more a hardware problem in your case than one on the software side.

mattdog01, 2020-09-10:
#19333 seems to have fixed this issue.  Very good stable printer operation. No random printhead movements indicative of SD read issues.  New Marlin GUI is so nice and works very well.  Will leave this issue open until it is tested in the main bugfix file with the mks_robin_stm32 HAL.  pins_MKS_ROBIN.h file was changed by @TheCodeExorcist and I used it to replace the pins_MKS_ROBIN.h file in one of the latest bugfixs (downloaded 20200909 evening).  Build enviro was mks_robin.  mks_robin_stm32 still needs work to make it stable. Thanks to all for your great work.

sjasonsmith, 2020-10-28:
@rhapsodyv have you changed anything on the mks_robin_stm32 environment that would have improved the SD issues @mattdog01?

I believe he is currently using mks_robin to avoid the issue, but it’s important for us to know if STM32F1 boards have reliable SD support with the STM32 environment as well.

mattdog01, 2020-10-28:
Thanks @sjasonsmith, Yes I'm still using the mks_robin enviro.  I the near future I will try a fresh download of bugfix and try the mks_robin_stm32 enviro again.  So far using the mks_robin environment with the updated robin_pin file it works perfectly.  I love that it shows the SD card active at first bootup and if I remove and reinstall the card (with the printer on, but not during a print) then it will ask if I want to refresh the connection if I go to start a print.  so nice. Prints are "perfect" and it seems that those pin definitions really did the trick for the mks_robin environment. I look forward to carving a couple hours out of the weekend to give time for printer play time.  Thanks so much everyone for your work on the Marlin project.  I just love having my machine running Marlin and all the great additions that have been possible using it.

mattdog01, 2020-11-02:
Hello whoever is attached to this bug. Tried a fresh download of bugfix today.  I keep getting an error in the build in the mks_robin_stm environment.  
Perhaps it is what I selected in the configuration file.  Here is the error that results in a build fail.

Archiving .pio\build\mks_robin_stm32\lib043\libIWatchdog.a
Archiving .pio\build\mks_robin_stm32\libbb1\libWire.a
Linking .pio\build\mks_robin_stm32\firmware.elf
c:/users/mcrawford/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/mks_robin_stm32/src/src/gcode/gcode.cpp.o: in function `GcodeSuite::process_parsed_command(bool)':
e:\Marlin Firmware\Marlin-bugfix-2.0.x 201101 New GUI/Marlin\src\gcode/gcode.cpp:558: undefined reference to `GcodeSuite::M145()'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\mks_robin_stm32\firmware.elf] Error 1

ellensp, 2020-11-02:
Is this the same config files as originally posted?

ellensp, 2020-11-02:
Made my own configs based on your original configs and updated to bugfix standards
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/5472509/configs.zip)
Compiles without issues.

sjasonsmith, 2020-11-02:
@ellensp are you compiling with mks_robin or mks_robin_stm32?

ellensp, 2020-11-02:
Included platformio.ini in configs in case of that question. mks_robin_stm32

mattdog01, 2020-11-03:
@ellensp wow that is good news. Thanks for letting me know. I will have to search through and see what I did incorrectly or didn't do correctly. I normal use the compare feature in PlatformIO to update the new config based on the old config. I won't be able to get to this right away but I will post my findings. Good to know it works though. Thanks.

sjasonsmith, 2020-11-16:
@mattdog01 any update on this? We really appreciate your willingness to help try things out as we prepare to move STM32F1 boards over to HAL/STM32.

mattdog01, 2020-11-17:
@sjasonsmith I'm so sorry that it has taken me so long to get back to this.  I just completed some testing.  I have some issues with the newest of new bugfix downloads.

1) No SD card selectability. Shows that is has been detected but can't be selected.
    *fixed added #define SDIO_SUPPORT under #define SDSUPPORT to configuration.h
    Needs to be refreshed to get it to read SD contents.

2) No sound on the buzzer.

3) Still have the random print head movements while printing (runs to the max or min end of the axis during the print) This indicates an issue with SPI stability reading the SD card.

4) The part cooling fan can't be controlled from the touch screen (G-code turns it on and it indicates correctly. If it is on and a different speed is selected, via the touch screen, it turns off)

5) During the bed level probing just before it moves, after the second probe, it raises a little more then moves. I have an offset of 2.15 so the clearance is around 7.15 well above the 5mm required. The display show Z at 9 and change when it is moving.
#define Z_CLEARANCE_DEPLOY_PROBE    5 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points
#define Z_CLEARANCE_MULTI_PROBE     3 // Z Clearance between multiple probes

6) Touches are not always being sensed by the touch screen.  I have single touch enabled.  Single touch needs to be repeated to get it to register or hold it there longer. 

Full disclosure I swapped the servo pins in the pins_MKS_ROBIN.h file so that I didn't need to go physically swap to the first servo spot.

[bugfix download 20201116 3hrs before this post.zip](https://github.com/MarlinFirmware/Marlin/files/5550912/bugfix.download.20201116.3hrs.before.this.post.zip)

I didn't continue with the print with the random print head movements.

Thanks for your continued work.
Let me know if you want me to do something specific.
Matt

mattdog01, 2020-11-29:
So I did another download of bugfix at 11:45 AM Eastern Standard Time today.  
Updated the config files and compiled in the mks_robin_stm32 environment.

1. No SD card selectability. Shows that is has been detected but can't be selected.
- still needed to add #define SDIO_SUPPORT under #define SDSUPPORT to configuration.h
- Still needs to be refreshed to get it to read SD contents, does not automatically see the SD card.

2. No sound on the buzzer.
- no change here, still no beeper. 
- tried to enable it in the config but that just had the cooling fan running at 100%
  uncommented these two lines in configuration.h (didn't think it would enable it but thought I would try)
  //#define LCD_FEEDBACK_FREQUENCY_DURATION_MS 2
  //#define LCD_FEEDBACK_FREQUENCY_HZ 5000
- is there some other place to enable the buzzer? (the old pins file had it defined and it worked)
 
3. Still have the random print head movements while printing (runs to the max or min end of the axis during the print) This indicates an issue with SPI stability reading the SD card.
**- watchdog stopped the print**

4. The part cooling fan can't be controlled from the touch screen (G-code turns it on and it indicates correctly. If it is on and a different speed is selected, via the touch screen, it turns off)
- this has been fixed by the updated bugfix and works just fine now (**thanks**)

5. During the bed level probing just before it moves, after the second probe, it raises a little more then moves. I have an offset of 2.15 so the clearance is around 7.15 well above the 5mm required. The display show Z at 9 and change when it is moving.
 #define Z_CLEARANCE_DEPLOY_PROBE 5 // Z Clearance for Deploy/Stow
 #define Z_CLEARANCE_BETWEEN_PROBES 5 // Z Clearance between probe points
 #define Z_CLEARANCE_MULTI_PROBE 3 // Z Clearance between multiple probes
- this has been fixed and bed level  functions are back to normal (**thanks**)

6. Touches are not always being sensed by the touch screen. I have single touch enabled. Single touch needs to be repeated to get it to register or hold it there longer.
- this has been corrected by the updated bugfix, works perfectly now. (**thanks**)

I moved the servo connector so I didn't need to change the pins file.
Here are the config files. 

[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/5611424/Configuration.files.zip)

 Let me know if there is something else I should try to help support the devs.
There has been progress and this is really great.
Thanks
Matt

mattdog01, 2020-11-29:
@sjasonsmith I'm not sure if you saw you update from yesterday.  This morning I tried to compile in the mks_robin environment and the screen is blank and without the beeper I can't tell if the touch screen is responding. I looked through the display settings and I think I have it right.  The only change I made was specifying the new build environment in platformio.ini

mattdog01, 2020-11-29:
The only bugfix that has been working well for me was downloaded 200909.  Does anyone know if any of the more recent versions have been working in the **mks_robin environment** or better yet the **mks_robin_stm32 environment**?
What is the goal? to get mks_robin_stm32 working well? 
Why is mks_robin not the desired environment? 
Does it have something to do with a more compatible HAL across the different 32 bit boards?
Has development of the mks_robin environment/profile stopped? (not even sure if I'm asking the right questions)

sjasonsmith, 2020-11-29:
I annoy sure about anything that would be preventing mks_robin from working right now.

We are trying to move away from the STM32F1 HAL because the STM32 HAL should support all STM32 MCUs with one HAL. The framework it uses is actually maintained by ST, whereas the F1 HAL uses the Maple framework which is maintained by a third party, and supports only a sliver of STM32 MCUs.

Maple support is also being deprecated in PIO, so it is beneficial to move to the framework with greater support.

@rhapsodyv are you aware of any current issues that would be impacting Robin boards working even in the F1 HAL? I was usi by F1 boards fine last night, although not Robin boards and without a TFT.

mattdog01, 2020-11-30:
@sjasonsmith Thanks for the info.  I will look and see if I have something incorrect in my configuration files.  (they are attached to the post from 2 days ago just in case someone wants to see what I did)
I love the methodology behind the one HAL for all STM32's.  I'm sure that is a bear from the dev side.  It is great that the different devs have found a way to provide the support for the different boards but I agree that this should eventually be pulled into the singular HAL support and then inside that framework provide the individual support to each of those individual boards.  Do I get the gist of the plan?
Let me know if I can provide any hardware help for testing or if you need me to do anything specific for the devs.  Thanks again.

sjasonsmith, 2020-11-30:
Yes, you are understanding the intent correctly. Unfortunately these boards are hard to locate, as MKS seems to be consolidating their product listings to favor E3 and Nano boards. rhapsodyv has it working well on Nano boards, but doesn't have this to test with. I managed to find one on AliExpress, it should arrive sometime in the next 3-4 years.

For now I think you should just focus on solving the issue impacting the mks_robin environment, and hold off on testing with mks_robin_stm32 until we make a more concerted effort to move boards over. I'd like to keep this issue specific to problems with mks_robin_stm32 , so you may need to report another issue if you can't figure out what is wrong with mks_robin.

mattdog01, 2020-12-05:
@sjasonsmith Hello Jason.  So I think I have some relevant info related to my UI issue in the **mks_robin eviro**.  Should this go to a different bug report or leave it here?

My screen is completely **blank** if I uncomment #define TFT_COLOR_UI but it works just fine if I uncomment #define TFT_CLASSIC_UI.  Touchscreen works in both configs but without visual reference in the blank screen.
I did minimal changes to the config files to exclude other factors. Only changed physical things in the configuration.h file (attached) no changes to configuation_adv.h.  All of this done with a fresh download of bug fix just a short while ago this morning.

/**
 * TFT UI - User Interface Selection. Enable one of the following options:
 *
 *   TFT_CLASSIC_UI - Emulated DOGM - 128x64 Upscaled
 *   TFT_COLOR_UI   - Marlin Default Menus, Touch Friendly, using full TFT capabilities
 *   TFT_LVGL_UI    - A Modern UI using LVGL
 *
 *   For LVGL_UI also copy the 'assets' folder from the build directory to the
 *   root of your SD card, together with the compiled firmware.
 */
//#define TFT_CLASSIC_UI
#define TFT_COLOR_UI
//#define TFT_LVGL_UI

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5647369/Configuration.zip)

mattdog01, 2020-12-05:
oops I just reread your post to " I'd like to keep this issue specific to problems with mks_robin_stm32 , so you may need to report another issue if you can't figure out what is wrong with mks_robin."
sorry off to find a suitable location.

rhapsodyv, 2020-12-09:
@mattdog01 this issue was fixed too, with our last fixes from #20404 ?

mattdog01, 2020-12-09:
@rhapsodyv It sure was.  and my other bug report too.  Gettin' it all cleaned up.

rhapsodyv, 2020-12-09:
So, for everyone following this thread: the problem was a bad assigned SD_DETEC_PIN ... it looked like a SDIO problem, because that bad pin assignment only happened when using SDIO....

@mattdog01, if it's working, you can close the issues. I will merge the fix now.

mattdog01, 2020-12-10:
All of the issues report in this bug have been resolved and the resolution has been merged.  Thank again folks for your work.  Hopefully this keeps the issue closed.

mattdog01, 2020-12-11:
I guess I was too hasty with closing this bug.  There has been some significant progress **but** using the mks_robin_stm32 environment I had the following problems after about 4 hours of printing.

1) evidence of another X axis run to max.  :pleading_face: 
2) The screen was also not reporting the fan speed. :cry: 
3) When I tapped the fan icon, to adjust the fan speed, the screen went blank. :sob:  
    leaving only the blue background and no text or icons.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5681805/Configuration.zip)
    the back light stayed on. :zany_face: 

This is from a download of bugfix Dec 9th configured fully and uploaded to the machine.  Used the exact same bug fix without any changes but built in the mks_robin environment and it has been running flawlessly for about 10 hours of printing.

tpruvot, 2020-12-12:
made two prints today on a similar printer (longer3D LK) from SD with color ui (first time)... no issues, but was just small fan ducts, one hour max. I already seen such move to X or Y max in the past, but was fixed long time ago... fan speed was on screen. (10 Dec build)

Could that be related to the "merge all constants" issue ?

rhapsodyv, 2020-12-28:
The SDIO improvements were merged. Can you test again?

mattdog01, 2021-01-01:
Sorry for the delay guys.  Downloaded the bugfix   #define STRING_DISTRIBUTION_DATE "2020-12-30" and ran a print.  So far, after about 8 hours of printing, everything seems to be fine and working well.  I'll keep going and let you know if anything crops up but it is looking good.  
I noticed one minor issue with menu navigation on the SD card.  It seems slower to respond than before, it takes a few moments for a tap to register on the display.  It almost seems that it wants double tap in the folders on the SD card even though double tap is disable.  Today for example...i was in /to Print/PETG White/Emilys Stuff/gate support.gcode.  Bit of a delay (1 second) into "To Print", a little longer delay (2 seconds) into "PETG White" the longest delay (4 ish seconds) into "Emilys Stuff". Then instant response to print "gate support.gcode".  I'll keep trying to see if this continues and see if I can find a pattern.

I noticed some nice fixes- tramming seems to work now with its error detection setup to show my probe offset was compatible with my inset area.  Fixed all that and worked great.  Man my bed sucks.  I made it from a bent 500 x 500 piece of 4mm aluminum. Tried to roll it flat but it is not perfect, but good enough.

mattdog01, 2021-01-03:
No need to keep this issue open as the original issue has been resolved for sometime and several iterations of the firmware.

PR #20404 (2020-12-09): MKS Robin TFT pins fixes for STM32 and STM32F1
### Description

Update the TFT pins for the MKS Robin board after the TFT refactoring.

Fixed a few more issues on this board, for STM32F1:

 - There's no SD_DETECT_PIN on MKS Robin V2.3 (Matt board). Only on > V2.4.
 - Fixed touch on STM32F1 too

### Benefits

Fix #20377
Fix #19274 

### Related Issues 

#15474
PR #20570 (2020-12-28): Better Defaults and Compatibility for SDIO in STM32
### Description

SDIO from STM32 have some different defaults and configurable options, from STM32F1. This PR add the compatibility options and match the default parameters.

### Benefits

Make STM32 more compatible with STM32F1, and more stable.

Should fix #19274
closed 2021-01-03 by mattdog01