MarlinFirmware/Marlin #19275: BLTouch variable probing speed during bed levelling

duceppemo, 2020-09-06:
### Description

In order to increase the probing speed but retaining high accuracy, it would be great to have multiple speeds on individual probings during the bed levelling procedure. After homing the Z, I think it is safe to think that the z position of the probing points shouldn't be higher than 1mm of the point used for the z homing.

Thus, in order to increase the speed for the auto bed levelling, it would be good to have portion of the z travel corresponding to (`Z_CLEARANCE_DEPLOY_PROBE` - 1) say at `Z_PROBE_SPEED_FAST` and the last 1mm at `Z_PROBE_SPEED_SLOW`.

### Feature Workflow

1. Modify the code to allow multiple speeds are various z heights during the auto bed levelling procedure. User would be able to set the height of the speed change.
2. Add extra variable(s) if needed.
3. Should not be available during z homing since bed height is unknown.

AnHardt, 2020-09-06:
It is exactly that way. (At least it was when i looked the last time)

> // Homing speeds (mm/min)
#define HOMING_FEEDRATE_XY (50*60)
#define HOMING_FEEDRATE_Z  (4*60)

> // Feedrate (mm/min) for the first approach when double-probing (MULTIPLE_PROBING == 2)
#define Z_PROBE_SPEED_FAST HOMING_FEEDRATE_Z

> // Feedrate (mm/min) for the "accurate" probe of each point
#define Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST / 2)

You just have to configure it to your taste. All speeds can be configured by a number. It does not need a relation to `HOMING_FEEDRATE_Z`

duceppemo, 2020-09-08:
Maybe I wasn't clear. I mean changing the speed during a single probe. Say the first 14mm done at 20mm/s and the last 1mm at 5mm/s (or whatever distance you choose.

AnHardt, 2020-09-08:
`Z_PROBE_SPEED_FAST` **is** the fast speed you are requesting.
`Z_PROBE_SPEED_SLOW` **is** the slower speed you are requesting.
Depending on the situation `Z_CLEARANCE_BETWEEN_PROBES` or `Z_CLEARANCE_MULTI_PROBE    ` **is** the height you are requesting.
What do you want moore?

thisiskeithb, 2020-09-10:
> `Z_PROBE_SPEED_FAST` **is** the fast speed you are requesting.
> `Z_PROBE_SPEED_SLOW` **is** the slower speed you are requesting.
> Depending on the situation `Z_CLEARANCE_BETWEEN_PROBES` or `Z_CLEARANCE_MULTI_PROBE ` **is** the height you are requesting.
> What do you want moore?

@duceppemo: Do these cover what you need?

ManuelMcLure, 2020-09-10:
I don’t think that those variables work the way @AnHardt is thinking. The slow speed is used for the second probe if probing multiple times. I believe what @duceppemo is asking for is for the speed to vary within a single probing move. So Z gets homed, the probe moves up 2mm or so at the slow speed, then the rest of `Z_CLEARANCE_BETWEEN_PROBES`, moves to the first point to be probed, moves down `Z_CLEARANCE_BETWEEN_PROBES` - 2mm at high speed and slows down to the slow speed for the last 2mm. 
This might eliminate the need for double probing altogether while still allowing for probes that need extra clearance to deploy to be used.

AnHardt, 2020-09-10:
Oh boys.
Think about what limits/criteria we have to adjust the parameters.

The requested height to switch from fast to slow? As deep as possible, but probe shall not be triggered then at any place of the area to probe.
`Z_CLEARANCE_BETWEEN_PROBES` ? As deep as possible, but probe shall not trigger at any point  on its way between the points to probe. (that's the same as above)
`Z_CLEARANCE_MULTI_PROBE` As deep as possible, but probe shall not be triggered at this probing point. (if this is a relative move it could be as small as the probes hysteresis - usually much smaller than the above)
All downward movements below this points have to be slow. All upward and above movement can be fast.
The difference between a first probe, after deploying the probe and the following ones is only - it comes with fast speed from `Z_CLEARANCE_DEPLOY_PROBE` what is usually higher than the other levels.
All moving before deploying does not belong to probing Z but is moving around in XY.

-----

~~One of the weaknesses of BL-Touches concept is - it has to be deployed for every single probe, so every single probe is a first probe. I don't care about BL-Touch and related code, for "opensource religious" reasons. All following additions like multiple probing, stallguard probing did not have a good theory. I gave up discussing the probe theory and code a long time ago.~~

You are right! 
After BL-Touch integration the probe code was destroyed beyond repair.

duceppemo, 2020-09-10:
@ManuelMcLure, that's exactly what I had in mind. That being said, I don't know how hard it would be to implement in the current code. I mean there are way to decrease auto bed levelling time by increasing speeds and reducing travel heights lie @AnHardt was describing. That being said, I think what I suggest would be an improvement of the probing workflow. Best of both worlds (speed and precision).