MarlinFirmware/Marlin #19277: [BUG] Z steppers get disabled when z-probe-offset-babystepping during printing

onfoot, 2020-09-06:
### Bug Description

Sometimes when modifying the Probe Z offset during printing the first layer, most likely when a retraction hop is happening simultaneously, the steppers in my independent dual-z setup get disengaged and there is no way of enabling them in any other way than by a reset.

### My Configurations

[stepper-bug-configs.zip](https://github.com/MarlinFirmware/Marlin/files/5179566/stepper-bug-configs.zip)

### Steps to Reproduce

1. Start an SD print
2. Watch in panic as nozzle is scraping the build plate or flying slightly above it.
3. Go into probe z-offset screen and start rotating the knob.
4. Hear the thump  of Z motors being disabled, emit a sigh of disappointment.
5. Stop the print.
6. Go into Motion/Move Axis/Z and try to move it around.
7. Notice no movement at all.
8. Be able to rotate the z-screws easily with fingers.

**Expected behavior:** The Z steppers are still powered and act on babystepping, perform retraction hops, and lift the nozzle (or drop the bed) to advance to subsequent layers.

**Actual behavior:** Z steppers are disabled and the z-screws can be moved freely, thus firmware has no control over them any longer.

#### Additional Information

I modified the MKS SGEN L v2 pin definition to enable the second Z stepper (I prefer to be able to autoalign the bed, but will eventually convert the printer to single Z stepper driving both screws using a belt).
[pins_MKS_SGEN_L_V2.h.zip](https://github.com/MarlinFirmware/Marlin/files/5180014/pins_MKS_SGEN_L_V2.h.zip)

sjasonsmith, 2020-09-06:
You might have more success with your 2209 drivers on the Z axis. They are less susceptible to this than 2208s.

You may also try enabling the experimental INTEGRATED_BABYSTEPPING.

onfoot, 2020-09-06:
Oh, so this is some kind of a driver issue?

I'll try the integrated babystepping, thanks for the pointer.

sjasonsmith, 2020-09-06:
Babystepping adds some unsynchronized stepper pulses into your normal sequence of steps. If your printer is already moving Z this can lead to an overload condition (or something) in TMC drivers, and they disable themselves to save themselves.

This came up before, but I was super busy at the time and didn't see exactly what the resolution was. You can probably search closed issues and find it, probably by searching for INTEGRATED_BABYSTEPPING.

Basically, some stuff was done to try to aid the situation, but I'm not sure how perfect the solution is. I haven't heard anyone mention it for a while.

onfoot, 2020-09-06:
Yes, found at least two, [this one](https://github.com/MarlinFirmware/Marlin/issues/16617) being the most interesting.

sjasonsmith, 2020-09-06:
OK, please try that out and report back how it works. Please don't immediately close this issue if it fixes things, because we may need to discuss how to make this more discoverable for people who need it.

onfoot, 2020-09-06:
Sure thing. I'll check it out as soon as I can.

onfoot, 2020-09-08:
After enabling integrated babystepping the issue is still there, I'm afraid. Most of the time when that happens both steppers get disabled. Sometimes, though, only one, which is always fun (and one more argument for moving to a single stepper setup). ;)

After I'm done printing a golden Mimikyu for my kid I'll try babystepping with StealthChop disabled, as the other issue I linked to suggested.

closed 2020-10-16 by github-actions[bot]