MarlinFirmware/Marlin #19278: [BUG] M104 with Smoothieboard goes into thermal runaway during bed leveling

timeshell, 2020-09-06:
### Bug Description

M104 with Smoothieboard goes into thermal runaway during bed leveling.

Essentially as it says.  I need to either do my bed leveling first and then head the hot end, or heat the hot end and wait for the hot end temperature to stabilize before starting bed leveling.  Otherwise the firmware doesn't seem to be able to get updates from the hotend indicating its temperature and goes into a thermal runaway protection error.

The smoothieboard uses LPC1769.  I don't seem to have this issue with my RE-ARM board, using LPC1768. 

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. M104 to heat the hotend
2. Start probing.
3. Rinse and repeat until you realize you need to do M109 instead.

**Expected behavior:** Hotend to heat successfully during probing.

**Actual behavior:** Thermal runaway occurs during probing.

ellensp, 2020-09-06:
Steps to reproduce are nonsensical
how do we replicate "Rinse and repeat until you realize you need to do M109 instead." 

Also the first smoothieboards from the kickstater were LPC1768, just to muddy the waters...

timeshell, 2020-09-06:
> Steps to reproduce are nonsensical
> how do we replicate "Rinse and repeat until you realize you need to do M109 instead."

Yah, point 3 was kind of supposed to be silly.  No offence intended.

> 
> Also the first smoothieboards from the kickstarter were LPC1768, just to muddy the waters...

This isn't one of the first Smoothieboards.  I kind of specifically mentioned that.  It is the V1.1 that uses the A5984 steppers.

timeshell, 2020-10-10:
Ping

boelle, 2020-10-13:
> Ping

please add some more meaningfull news.... "ping" is not very helpfull

timeshell, 2020-10-13:
Ok, my news is as far as I know it hasn't been addressed yet.  Therefore I'd like it to remain open.

boelle, 2020-10-13:
you probe with everything hot.... i assume if you do it cold then its does not error out

what during normal printing?

timeshell, 2020-10-16:
It works fine if I wait for it to heat up before probing or if I heat up after probing.  It goes into thermal runaway during bed leveling if I heat using M104 and do a bed leveling during the M104.  I do this using a RE-ARM board without any problems.  However it doesn't seem to work correctly on the Smoothieboard 5xC V1.1.

closed 2020-11-23 by github-actions[bot]