MarlinFirmware/Marlin #19286: x and y axis moving in wrong direction after homing

Future124, 2020-09-07:
I am having same issue with skr mini e3 v2.0 with built in tmc2209 drivers, when homing a second time or when going to auto bed level after homing, x and y axis start from where z safe homing position is at the center of my bed (150,150) thinking its a 0,0 position and crashes as it thinks it at 0,0 when it starts to auto bed level but its actually at 150,150. the x y and z motors all go in the proper direction as directed, its only after homing the first time and either trying to home a second time or autp bed lever right after homing. Another thing i did notice but not sure if its connected is if i manually move the motors to their 0,0 positons, they home properly the first time but still crash if homed a second time or if you auto level , however, lets say that they are at the 150,150 positions and i try and home them  for the first time they will also try and home in the opposite direction as well and crash. ive flipped the plugs for the drivers replasced the endstops, replaced the stepper motors tried bugfix build, the marlin that came with the board, and its always the same, if you need any more info please let me know, thank you.

Future124, 2020-09-07:
sorry first time opening issuee i am refrencing this issue that was closed : https://github.com/MarlinFirmware/Marlin/issues/16323

ellensp, 2020-09-07:
Please follow the template and provide Configuration files.

Future124, 2020-09-07:
ok ill be adding the zip file shortly https://github.com/MarlinFirmware/Marlin/issues/19289

thisiskeithb, 2020-09-07:
Don’t open multiple reports for the same issue. You can attach your configs to this issue in your next reply or edit your post and add them there.

Future124, 2020-09-07:
my bad wasnt intentional, waiting to see if my pc is going to restore as it crashed so i can get the congfig files

ellensp, 2020-09-08:
Still waiting for these config files....

ellensp, 2020-09-09:
no configs, closing (again)

closed 2020-09-09 by ellensp