MarlinFirmware/Marlin #19291: [BUG] Ender 3 does not know what the middle of the bed is.

Dutchone, 2020-09-07:
### Bug Description

Ender 3 with SKR E3 Mini V1.2 and 3.1 Bltouch with latest stable and bugfix 09-07-2020

Ender does not use the correct center of the bed.

### My Configurations

[Config 09-07-2020.zip](https://github.com/MarlinFirmware/Marlin/files/5183210/Config.09-07-2020.zip)

### Steps to Reproduce

1.  I have used M502 and M500 to try and resolve the issue.

2. Adjusting the bed back to 220x220 has the same affect. Still of center.

3. using the level corner function to measure what the printer sees as the middle of the bed.

**Expected behaviour:** [What you expect to happen]

With the bed set to 235x235 the center of the bed should be around  X=117.5 and Y=117.5, for probing and printing.

**Actual behaviour:** 

The printer bed has been set in config to 235x235. Measured with the left corner down as X=0 and Y= 0 is get the following measurements X= 120 and Y=104 when using level corners to move the nozzle to the center of the bed

When using level corners its about 1.4 cm from the front of the bed and 4cm at the back of the bed

#### Additional Information
[image]

[image]

thisiskeithb, 2020-09-07:
This is not a Marlin bug. Both your `X_MIN_POS` & `Y_MIN_POS` are 0 in your config, but you will need to find the correct values for your printer since it homes off the bed.

---
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-09-07 by thisiskeithb

Dutchone, 2020-09-07:
Strange, never had to do that before.

I will try that