MarlinFirmware/Marlin #19293: FAST_PWM_FAN affects serial communication on SKR 14.4

Quas7, 2020-09-07:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2020-09-07:
Closing. No content.

Quas7, 2020-09-07:
@sjasonsmith you are a bit too fast ;)

AnHardt, 2020-09-07:
Switching any power causes ESD. 
Adding a comment for a fan doesn't make much sense. We had to add the same warning for heaters and steppers. But do we really have to? It should be general knowledge.

Quas7, 2020-09-07:
@AnHardt 
I think, you mean EMI (electromagnetic interference) and not ESD (electrostatic discharge). 
But yes, any fast switching PWM will generate electormagnetic noise and a higher PWM frequency will add more noise. 
But, I would not really call this general knowledge (=highschool level?). ;)

I think, the case here is a bit special, because the LPC1769 with a 120Mhz clock seems to generate a electromagnetic radiation that affects the serial communication quite drastically. The same setting did not result in an issue for due+radds (84Mhz) or my old atmega boards (16Mhz) but with the SKR 1.4 turbo it fails on serial communication almost instantly.

Assuming the implementation is still following the explanation in line 412 pp. in Config_adv we get: 
F_PWM = F_CPU/(2*255*1) = 235.3 kHz

And this is more than 3x the "normal" range listed and it is well above 120-150kHz RFID frequencies (10cm range).
I will try to set it to 80kHz with #define FAST_PWM_FAN_FREQUENCY next and maybe test my nanoVNA for the first time.

I would aruge, that nobody needs a FAST_PWM above audible levels. 
Maybe it would make sense to default it to max 40kHz? That would be a dog safe level. ;)

Quas7, 2020-09-07:
alright, My problem is solved with FAST_PWM_FAN and #define FAST_PWM_FAN_FREQUENCY 30000 (pre-scaler 4).

my proposal for the 5 yr old remark on FAST_PWM_FAN:
// Increase the FAN PWM frequency. Removes the PWM noise but increases heating in the FET/Arduino.
// On high speed CPUs restrict to PWM frequency to reasonble limits (e.g. 40kHz) via #define FAST_PWM_FAN_FREQUENCY 
// Configuration_adv.h to avoid electromagnetic interference and possibly serial communication and other issues

AnHardt, 2020-09-08:
Sorry. Yes EMI. My fingers just made something existing from your EMC above.

Are we talking about a timer clock frequency x (after prescaler), or a PWM-pulse-rate (x/resolution)?

Quas7, 2020-09-08:
I think, we are talking about the PWM-pulse-rate, if I look at the equation in the adv_config as there is division by 2 that does not really fit normal timer scaling as I know it. (I edited the equation to be readable above)

AnHardt, 2020-09-08:
https://github.com/MarlinFirmware/Marlin/blob/9ade72aee94981b26be0ca125ab487166d0bd7e3/Marlin/src/inc/Conditionals_post.h#L2233-L2238

Is the source of the evil.
This really should not have F_CPU in, resulting in a factor eleven difference depending on if the processor runs on 16MHz or 168MHz. A fixed number would give much less surprising results.

And splitting the the definition for FAST_PWM_FAN from its FAST_PWM_FAN_FREQUENCY into a different file with unsensible default makes it worse. The comments will rarely be read.

Quas7, 2020-09-08:
Agreed. 
An upper limit should be applied to the frequency or at least a sanity check for >30 kHz.

On top, it might even be possible to brick a newer generation board while programming over serial connection via bootloader, if this high EMI is present.

closed 2020-10-28 by github-actions[bot]