MarlinFirmware/Marlin #19296: Marlin 2.0.6 autostart not working properly

3dseawa, 2020-09-07:
### Bug Description

Marlin 2.0.6 autostart not working properly.
I have three files sliced in Simply3D and renamed to auto0.g, auto1.g, auto2.g.  When each file is named auto0.g and is the only file on the SD card each will autostart upon power boot-up.  When auto0.g and auto1.g is on the card the auto1.g will autostart and print but not auto0.g.  When auto0.g, auto1.g, auto2.g are on the SD card nothing happens.  I have tried each of the files in different combinations with the same results.  
When the same three files are on one SD card and put into my Ender3 Pro the auto0.g will execute but the other two do not.  

The above leads me to believe this is a bug.  If it is a configuration or file setting issue, please let me know the proper configuration or file settings.

I am do not have a LCD connected, nor is the USB connected to a computer – just the motherboard so printing directly from the micro-SD card reader on the mother board.

I have a custom-built machine using Marlin 2.0.6, download on 9/6/2020, on a BigTree Tech SKR 1.4 Turbo with TMC2209 drivers and sensorless homing. These are the setting in Configuration_adv.h folder: (lines 1145 and 1147)
#define SD_MENU_CONFIRM_START             // Confirm the selected SD file before printing
#define MENU_ADDAUTOSTART               // Add a menu option to run auto#.g files 
No changes have been made to any .cpp files.
I have read others are having the same issue with auto1.g executing but not auto0.g.
Thank you in advance for helping identifying and resolving this issue for myself and other.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

Attached are the configuration.h and configuration_adv.h zipped files:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5184655/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5184657/Configuration_adv.zip)

### Steps to Reproduce

1. save three sliced .gcobe files and rename as auto0.g, auto1.g, auto2.g, 
2. load onto sd card reader on motherboard while disconnected from LCD and computer
3. power on machine and note what happens.

**Expected behavior:** [What you expect to happen]
upon power bootup auto0.g should  print, then auto1.g should print automatically, then auto2.g should print - up to auto9.g.  

Attached are the configuration.h and configuration_adv.h zipped files:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5184655/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5184657/Configuration_adv.zip)

**Actual behavior:** [What actually happens]
See above for what happens and what has been tried.
#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thomas374b, 2020-09-09:
Enable SDCARD_SORT_ALPHA in  Configuration_adv.h  and tried to play with optional defines around there.

This may solve your problem with starting auto1.g instead of auto0.g  since files are usually unsorted written to the card

thomas374b, 2020-09-09:
If you have defined SD_MENU_CONFIRM_START

you must press a button on the LCD in order to start printing. If you have no LCD this code will never be reached.

3dseawa, 2020-09-09:
Thank you thomas374b for you input.
I commented with // for SD_MENU_CONFIRM_START
I also uncommented (deleted the //) SDCARD_SORT_ALPHA 
Sadly after trying several tests , I got the same results as before - auto1.g executes but not auto0.g when those two files are on the SD card.  When auto0.g, auto1.g and auto2.g are all on the SD card then nothing executes. auto0.g will execute when it is the only file on the card.

AnHardt, 2020-09-09:
The riddle has likely to be solver [here](https://github.com/MarlinFirmware/Marlin/blob/9ade72aee94981b26be0ca125ab487166d0bd7e3/Marlin/src/sd/cardreader.cpp#L676-L699) or in the initialization of `autostart_index`.
At the moment i can't find a obvious problem - but i currently don't understand every part of the code properly.

boelle, 2020-09-10:
@3dseawa please check what @AnHardt said about https://github.com/MarlinFirmware/Marlin/blob/9ade72aee94981b26be0ca125ab487166d0bd7e3/Marlin/src/sd/cardreader.cpp#L676-L699

also Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

3dseawa, 2020-09-11:
Thank you AnHardt and Bo,
I looked at the code and it is beyond my limited coding knowledge.
It matches what I am using which is Marlin 2.0.6 Bugfix downloaded 9/6/2020.

thinkyhead, 2020-09-11:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

3dseawa, 2020-09-11:
I tested the autostart feature with the bugfix-2.0.downloaded today 9/11/20
branch based on the link provided by Scott - thank you Scott.
I encountered the same issues. As a reminder, I am printing directly from
the SD card on the Bigtree Tech SKR V1.4 board with TMC2209 drivers with
sensorless homing. I do not have an LCD or computer connected.
When auto0.g is the only file on the SD card it will execute.  When an
auto0.g file is on the SD card with .gco files auto0.g will execute.
When auto0.g and auto1.g are on the SD card the auto1.g only executes.
When auto0.g, auto1.g, and auto2.g are on the SD card nothing executes.
I test each of the three files as auto0.g and as auto1.g with the same
results.
Attached are the configuration.h and configuration_adv.h along with the
three very simple .g files I am testing.
I have also tested actual 3D printable files sliced in Simplfy3D with
the same results.

This problem is beyond by coding skills so I appreciate all of your help
and support resolving this issue for our community.

qwewer0, 2020-09-11:
Here is what is happens for me:

- **auto0.g**: Printer starts -> (LCD shows AUTO0.G) auto0.g runs -> Steppers gets disabled -> Other print jobs can be started
- **auto0.g** and **auto1.g**: Printer starts -> (LCD shows AUTO1.G) auto0.g runs -> Steppers gets disabled -> auto1.g runs -> Steppers gets disabled -> Other print jobs can be started
- **auto0.g**, **auto1.g** and **auto2.g**: Printer starts -> (LCD shows AUTO0.G) -> No gcode is executed, and the printer stucks in printing mode
After print is aborted, no other print (.gcode, .g etc.) can be started, I mean by that the printer won't accept the selected "Print" menu item in the print confirmation menu, can only clicked on "Cancel", and no Auto Home or M48 Probe Test can be run from the LCD. While it is in the printing mode or aborted it, wasn't able to connect it to a PC. Restore Defaults, Store Settings doesn't helps.

<details>
  <summary>auto0.g content</summary>
  
  ```
  G28
  ```
  
</details>

<details>
  <summary>auto1.g content</summary>
  
  ```
  G28
  M300 S2000 P5
  ```
  
</details>

<details>
  <summary>auto2.g content</summary>
  
  ```
  G28
  M300 S2000 P5
  G4 P50
  M300 S2000 P15
  ```
  
</details>

Here are the files:
[auto.zip](https://github.com/MarlinFirmware/Marlin/files/5211234/auto.zip)

Using the latest Bugfix-2.0.x
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5211268/Configuration.zip)

qwewer0, 2020-10-12:
Tested https://github.com/MarlinFirmware/Marlin/issues/19296#issuecomment-691283976 on the latest Bugfix-2.0.x, and the results are still the same.

3dseawa, 2020-10-16:
It looks like there is a bug with the Autostart feature and it is beyond my knowledge and skill set to fix.
I did find a workaround by only having auto0.g on my SD card (no other auto1.g, auto2.g, et.c. files) then using g-code M32 as the last line of the auto0.g  g-code file and naming the next file to start within the M32 command.  Then using M32 command in the second file by putting the M32 command as the last g-code line in that file's g-code and naming it the third file to start, and so on up to seven additional file.  My setup will only allow for a total of 8 files on the SD card for the auot0.g to execute. More than 8 files and auot0.g does not execute.  Not sure why it is limited to 8 files for the auto0.g to execute.  

A few key points about the M32 command:
1) the file name can be a maximum of 8 characters and the file type can only be 3 letters long -'filename.gco'. 
2) the character ! has to directly precede the file name and # must end the file name
e.g. M32 !filename.gco#
3) if you want to create a loop where the auto0.g will start after the last file then use the M32 command in the last line of the last file and name it: M32 !auto0.g# and it will execute the auto0.g file again and restart the series of files.
https://marlinfw.org/docs/gcode/M032.html
Thanks you to all who contributed to this issue.  I appreciate your efforts.

Zumili, 2020-11-07:
Hey guys, just found some possible solution.
Just change this line: https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/sd/cardreader.cpp#L708
from
`if (autostart_index < 0 || flag.sdprinting) return;`
to
`if (autostart_index < 0 || flag.sdprinting || queue.has_commands_queued()) return;`
on line 708 in cardreader.cpp

qwewer0, 2020-11-07:
Tested https://github.com/MarlinFirmware/Marlin/issues/19296#issuecomment-691283976 on the latest Bugfix-2.0.x with @Zumili fix https://github.com/MarlinFirmware/Marlin/issues/19296#issuecomment-723446341, and it seems to fixed the issue.
Here is what is happens for me:

- **auto0.g**: Printer starts -> (LCD shows AUTO0.G) auto0.g runs -> Steppers gets disabled -> Other print jobs can be started
- **auto0.g** and **auto1.g**: Printer starts -> (LCD shows AUTO0.G) auto0.g runs -> Steppers gets disabled -> (LCD shows AUTO1.G) auto1.g runs -> Steppers gets disabled -> Other print jobs can be started
- **auto0.g**, **auto1.g** and **auto2.g**: Printer starts -> (LCD shows AUTO0.G) auto0.g runs -> Steppers gets disabled -> (LCD shows AUTO1.G) auto1.g runs -> Steppers gets disabled -> (LCD shows AUTO2.G) auto2.g runs -> Steppers gets disabled -> Other print jobs can be started

So it works, and now it shows the right *.g file that is running.

@Zumili You should create a PR with that fix, or you could let me do it.

Zumili, 2020-11-07:
qwewer0 I am not very familiar with github and I don't want to  break the universe :) , so it will be nice if you can do it.

boelle, 2020-11-07:
@3dseawa remember to close this one when the above PR is merged (it goes purple color)

thinkyhead, 2020-11-08:
I've put together a comprehensive patch at #20071. Please test to confirm that those changes work for you.

3dseawa, 2020-11-08:
Thank you.
I will try to tested it the first part of the week.

boelle, 2020-11-22:
@3dseawa did you get time to test?

3dseawa, 2020-11-22:
No, sadly I had a home issue to repair which captivated my time.

3dseawa, 2020-11-22:
Firsty, thank you to all who have worked on this issue.
To test this complete fix/ comprehensive patch, I am assuming I clink on
the "20071" link from Scott Latheine dated 11/7 above then click on the '
MarlinFirmware:bugfix-2.0.x
<https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x>  ' link just
below the title tile of the 20071 link and then download that version in
that link and use that for my test as the referenced  MarlinFirmware:
bugfix-2.0.x <https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x>
has the fixes incorporated into that download?

Please let me know if this is correct or if there is something else I need
to do.  I should have my house put back together today and hope to work on
it this afternoon or evening.

closed 2020-12-07 by thinkyhead

qwewer0, 2020-12-07:
@thinkyhead Thank you for the AutoStart refractory!

On the latest bugfix-2.0.x (with https://github.com/MarlinFirmware/Marlin/pull/20071) [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5652546/Configuration.zip)

Here is what is happens:

- **auto0.g**: Printer starts -> (LCD shows AUTO0.G) auto0.g runs -> Steppers stays engaged -> **Printer is stuck in printing mode** -> After print is aborted other print jobs can be started
- **auto0.g** and **auto1.g**: Printer starts -> (LCD shows AUTO0.G) auto0.g runs -> Steppers gets disabled -> (LCD shows AUTO1.G) auto1.g runs -> Steppers stays engaged -> **Printer is stuck in printing mode** -> After print is aborted other print jobs can be started
- **auto0.g**, **auto1.g** and **auto2.g**: Printer starts -> (LCD shows AUTO0.G) auto0.g runs -> Steppers gets disabled -> (LCD shows AUTO1.G) auto1.g runs ->Steppers gets disabled -> (LCD shows AUTO2.G) auto2.g runs -> Steppers stays engaged -> **Printer is stuck in printing mode** -> After print is aborted other print jobs can be started

With or without auto#.g files, **a started print job stays in printing mode.**
With `NO_SD_AUTOSTART` disabled, there is no print job issue.

qwewer0, 2020-12-08:
@3dseawa Were you able to try it?

3dseawa, 2020-12-08:
Not yet. My board died and replacement is on its way.

3dseawa, 2020-12-27:
For some reason my g-code program stops a few lines from the end of the
program so it does not advance to the next program.
This has me stumped. I will  keep you posted what I find out.

HarpoDog, 2021-02-11:
I am please to confirm that the autostart feature is now working as designed.
Thank you to the entire team that resolved this issue.
You are truly appreciated!

3dseawa, 2021-02-11:
Yes, Thank you to all that resolved this issue.
Mine is now working and your efforts are appreciated.