MarlinFirmware/Marlin #19302: Inactivity Shutdown triggered during filament change

SteveDC2, 2020-09-08:
If the inactivity timer is enabled with M85 and a filament change is triggered due to filament running out then the printer will KILL after the inactivity time.
The hot end looks like it was disabled but the bed remained enabled when the filament ran out.

Obviously this should not happen.

I would expect the hot end to remain active for whatever period is set in FW followed by a cooldown and re-heat as necessary, but with the inactivity timeout disabled.

ellensp, 2020-09-08:
Please follow the template and attach you Configuration files.

SteveDC2, 2020-09-08:
Here are my configuration files. The print GCode executes M85 S120 to turn on the activity timer at the start of the print.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5190676/Marlin.zip)

SteveDC2, 2020-09-09:
Same thing just happened to my other printer.
This seems to occur on both 1.1.9 and 2.0.x
Attached are my Marlin 2 files.
[Marlin20x.zip](https://github.com/MarlinFirmware/Marlin/files/5191955/Marlin20x.zip)

ellensp, 2020-09-09:
@SteveDC2 you first configs show your using marlin 010109.
This is no longer being developed.
The second config shows 020005 not the current marlin.

Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

SteveDC2, 2020-09-09:
I am currently using the bugfix code base but I copied my config files over from my previous 2.0x build.

I just compared the bugfix templates against my config files and the only place I see differences potentially something related to temp and timeout is around line 346 in configuration_adv.h but this seems more related to generic timeout of the hot end and not the inactivity timer.

I will migrate my config settings to the 'new' template and check again but I couldn't see any significant differences.

BR,
Steve

SteveDC2, 2020-09-09:
Just to add... I have just experienced the KILL again on Marlin 2 bug fix. I was waiting for the filament to run out and immediately tried to swap it out before the timeout but didn't get the print restarted in time so this is definitely related to a filament change.

SteveDC2, 2020-09-10:
Just done a completely clean Marlin 2 bugfix build and don't seem to see the issue anymore (using my original configs with minor changes).

I guess I wasn't using the bugfix build before afterall.

I will continue to keep an eye on the behavior to make sure it isn't just a timing thing that allowed the change to occur without KILL.

Would be nice to back port whatever the change was to Marlin 1.1.9 since it certainly exists there. Any pointers on what where best to look in the code for where this was fixed so I might back port the changes to 1.1.9? 

I guess for now this report can be closed.

Regards,
Steve

closed 2020-09-10 by boelle