MarlinFirmware/Marlin #19305: [BUG] Compilation error, missing "libmaple/usart.h"

marciot, 2020-09-08:
On the upstream/bugfix-2.0.x branch, even when not compiling for boards other than STM32F1, I get the following error:

```
src/HAL/STM32F1/MarlinSerial.cpp:25:28: fatal error: libmaple/usart.h: No such file or directory
compilation terminated.
Makefile:874: recipe for target 'applet/src/HAL/STM32F1/MarlinSerial.o' failed
make: *** [applet/src/HAL/STM32F1/MarlinSerial.o] Error 1
m
```

It seems like this may be missing a guard of some sort.

rhapsodyv, 2020-09-08:
Are you using VSCode?

If you are not compiling for STM32F1, the folder `src/HAL/STM32F1` will not enter in the compilation process, when using vscode+platformio.

marciot, 2020-09-08:
No, I am using the Makefile. It looks like in other files this is resolved using "#ifdef __STM32F1__". I'll make the necessary changes in a PR.

rhapsodyv, 2020-09-08:
If you are using custom Makefile, you can just skip all files that aren't for your target architecture. PlatformIO is doing this. The compilation process is a lot faster this way.

marciot, 2020-09-08:
@rhapsodyv: You make a good point. But that tweak to the Makefile for the moment is more than I know how to do. I'll just go ahead with the #ifdef for now since it follows the precendent that was set before.

rhapsodyv, 2020-09-08:
Why not use just `platformio run` to build from command line? 
You can pass defines for it.

Look at this script that I created to build all chitu variations: https://github.com/rhapsodyv/Marlin/blob/tronxy-full-mks-tft/build-all-chitu.sh

marciot, 2020-09-08:
@rhapsodyv: I actually do this for certain boards for which the Makefile does not work. I just haven't been able to switch over completely for the majority of my builds. IIRC, I ran into problems with a platfromio build for certain printers and I just didn't have time to troubleshoot it.

So for the moment I've taken the responsibility of making sure the Makefile continues to work :)

I made a PR: https://github.com/MarlinFirmware/Marlin/pull/19307

closed 2020-09-08 by marciot