MarlinFirmware/Marlin #19306: [BUG] (Command M701, M702 does not work correctly.)

user32rtu, 2020-09-08:
**Command M701, M702 does not work correctly.**

_Sapphire Pro printer. MKS Robin Nano v1.2 board._

Loading / unloading filament does not work correctly, as well as commands M701, M702.
Previously, I used firmware from the le3tspeak / Marlin-2.0.X-MKS-Robin-Nano branch and everything worked correctly. 
I decided to try the firmware from Marlin / bugfix-2.0.x. 
Moved settings from le3tspeak / Marlin-2.0.X-MKS-Robin-Nano firmware to Marlin / bugfix-2.0.x.
After downloading the firmware to the printer, I found two errors:

 1. movement along the XY axes has become incorrect. Changed “INVERT_X_DIR” to true in configuration.h, after which the  direction of movement became correct.

 2. When giving the command to load / unload from the printer menu, something incomprehensible happened. It seems that instead of moving the extruder, there is a movement along the Z axis. This can be seen in the video.
https://yadi.sk/i/Tn2FToKqmOgH4Q
https://yadi.sk/i/lEf6MdFtYMQUiA

The same happens when sending commands M701, M702.

GMagician, 2020-09-10:
What I see is a z raise. AFAIK filament change move to park position and may rise Z. After this I see a "heating" state then you have to wait temp is reached before "load" or "unload" may be done

qwewer0, 2020-09-10:
@user32rtu In your Configuration.h, you have Z_raise set to 50, so that is the Z axis movement in your video.

boelle, 2020-09-10:
@GMagician & @qwewer0  so this is most likely a user config error or misunderstanding?

qwewer0, 2020-09-10:
@boelle I think everything is working as it is intended. So yes, user config error and misunderstanding.

boelle, 2020-09-10:
oki, will close then, it can always be reopned if its a true bug

user32rtu, 2020-09-11:
> What I see is a z raise. AFAIK filament change move to park position and may raise Z. After this I see a "heating" state then you have to wait temp is reached before "load" or "unload" may be done

I thought the video had enough information. But apparently I was wrong. I'll describe what I see. 
The second video shows that the hotend has warmed up to the operating temperature and instead of unloading the filament, the table begins to rise. After that, the operation is completed.

user32rtu, 2020-09-11:
I thought the video had enough information. But apparently I was wrong. I'll describe what I see.
The second video shows that the hotend has warmed up to the operating temperature and instead of unloading the filament, the table begins to rise. After that, the operation is completed.

qwewer0, 2020-09-11:
Try to put an M25 somewhere at the start after homing in a gcode file, then see if it works as intended. If not, then maybe https://github.com/MarlinFirmware/Marlin/issues/18191 is the issue.

GMagician, 2020-09-11:
Not clear what I see after heated. It seem something happens on extruder, maybe unload?

qwewer0, 2020-09-11:
> The second video shows that the hotend has warmed up to the operating temperature and instead of unloading the filament, the table begins to rise. After that, the operation is completed.

This is what unloading does for me:
Select Unload Filament -> Select Temp setting ->Print head raises and the nozzle heating begins ->Temperature reached -> Unloads filament -> Print head sinks back to the original position with the heating still on.

@user32rtu Do you have `INVERT_E0_DIR` set correctly?

user32rtu, 2020-09-11:
> @user32rtu Do you have `INVERT_E0_DIR` set correctly?
Yes. It has always been this way and worked right.

user32rtu, 2020-09-11:
> This is what unloading does for me:
> Select Unload Filament -> Select Temp setting ->Print head raises and the nozzle heating begins ->Temperature reached -> Unloads filament -> Print head sinks back to the original position with the heating still on.
> 
Everything works exactly the same with the le3tspeak firmware.

user32rtu, 2020-09-11:
Now updated PlatformIO and Python. I rebuilt the firmware and it worked.
It is not clear what happened.
Thank you all for your assistance.
The topic is closed.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5209302/Marlin.zip)

closed 2020-09-11 by user32rtu