MarlinFirmware/Marlin #19308: Double Z Endstop problem

Kyosuke33, 2020-09-08:
Hello, i have the same problem as the issue #12081 closed apprently without solution since 2018.

I have a Sapphire Plus Core Xy with MKS SGEN L v2 and two endstop on Z.

The problem is that the dual Z endstop won't work. 

Let say Z1 is left (plug on Zmin) and Z2 is right (plug on Zmax).

When i try to home Z axis, there is two case :  
- Z1 reach the endstop first until it's triggered, and all stop, Z2 motor stop spin before endstop triggered. If i touch the Z2 endstop, all bump back and Z1 reach the endstop again and home if finished.
- Z2 reach the endstop first but won't stop when touch the endstop and force against the endstop until Z1 triggered the endstop, all bump and home is finished.

I try to switch motors, câble, endstop but nothing work.

https://youtu.be/CB-Q-3VNvFY
https://youtu.be/snWqHsrwZ1s

thisiskeithb, 2020-09-08:
Please follow the Issue Template when filing a bug report. This includes attaching your configs:

```md
<!--

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/ or the Marlin Discord Server https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

<!-- Description of the bug or requested feature -->

### Steps to Reproduce

<!-- If this is a Bug Report, please describe the steps needed to reproduce the issue -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
```

Kyosuke33, 2020-09-08:
Sorry, i try to add .rar but won't work. i just add my config .zip

Kyosuke33, 2020-09-08:
I was thinking that maybe the problem is the diag pin on the TMC2209 driver, but X and Y work great without removing them. (i use them on Standalone mode)

GMagician, 2020-09-08:
Check what M119 says when you press Z1 and then Z2 or Z2 and tren Z1

Kyosuke33, 2020-09-08:
[image]

Kyosuke33, 2020-09-08:
Closed issue, i'm a noob xD there is 2 Z plug on the board but one driver. I was thinking there is two for Z... I plug on the E1 and it work. Thank you.

closed 2020-09-08 by Kyosuke33