MarlinFirmware/Marlin #19310: [BUG] M115 does not return STRING_CONFIG_H_AUTHOR

ChrisToxz, 2020-09-08:
### Bug Description

M115 should return `STRING_CONFIG_H_AUTHOR`, but doesn't.

As explained in the config:
`// Author info of this build printed to the host during boot and M115`

And explained on the website: https://marlinfw.org/docs/configuration/configuration.html
`STRING_CONFIG_H_AUTHOR is shown in the Marlin startup message, and is meant to identify the author (and optional variant) of the firmware. Use this setting as a way to uniquely identify all your custom configurations. The startup message is printed when connecting to host software, when the board reboots and M115.`

### My Configurations
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5191134/Marlin.zip)

### Steps to Reproduce

1. Send M115 command

**Expected behavior:**
Show `STRING_CONFIG_H_AUTHOR` as well

**Actual behavior:**

```
Send: M115
Recv: FIRMWARE_NAME:Marlin 2.0.6 (Sep  5 2020 20:59:07) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Toxz Ender EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFLEVEL:0
Recv: Cap:RUNOUT:1
Recv: Cap:Z_PROBE:0
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:PROMPT_SUPPORT:0
Recv: Cap:SDCARD:1
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:LONG_FILENAME:0
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:1
Recv: Cap:BABYSTEPPING:0
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: ok
```

#### Additional Information

Cannot find any reference to STRING_CONFIG_H_AUTHOR in m115.h and language.h

thisiskeithb, 2020-09-08:
The "and M115" is probably a typo or Marlin `2.0.x` no longer shows this information in `M115`. I also checked Marlin `1.1.x`, which has been EOL'd, but it only shows on boot as well.

I created a PR in the Marlin Documentation repo to fix this reference: https://github.com/MarlinFirmware/MarlinDocumentation/pull/334

ChrisToxz, 2020-09-09:
Also it isn't showed at boot (at least when host connects)
`The startup message is printed when connecting to host software`
When connect it prints M115 but without `STRING_CONFIG_H_AUTHOR`

Why not adding this to M115, it can be very useful to check current config.

thisiskeithb, 2020-09-09:
> Also it isn't showed at boot (at least when host connects)

This won’t show when your host connects, but on board power up. Power cycle your board with host that auto-reconnects like Arduino IDE’s serial monitor or minicom and you will see that it does display `STRING_CONFIG_H_AUTHOR`.

Related code:
https://github.com/MarlinFirmware/Marlin/blob/bc7720c0cd3917f44200c0b78e1b635e4c7b6797/Marlin/src/MarlinCore.cpp#L981-L988

ChrisToxz, 2020-09-09:
In your PR the following is written:
`The startup message is printed when connecting to host software and whenever the board reboots.'

I understand from connecting to host software that this is when serial is connected? For e.g. Octoprint. It shows just M115.

I still do not understand why not to add this into M115?
" M115 - Print the firmware info ".
The config author is imo a part of the firmware info, at least usefull.

thinkyhead, 2020-09-10:
As innocuous as the change may be, we have to coordinate any proposed changes to the output of `M115` with the developers of host software to make sure that their code can tolerate the change, and to settle on a standard label for the new entry. Once applied, the change needs to be documented in the RepRap wiki.

closed 2020-09-10 by thinkyhead