MarlinFirmware/Marlin #19313: [BUG] M502/M500 when TMC drivers do not have VMOT results in incorrect values being stored in EEPROM

ManuelMcLure, 2020-09-09:
### Bug Description

If a motherboard is powered by USB but the main power supply is not powered on, executing `M500` will save the wrong values into EEPROM.

### My Configurations

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/5192308/Config.zip)

### Steps to Reproduce

* Use my configuration on an LPC-1768-based board.
* Power board with main power.
* Execute `M502`.
* Execute `M503` and see that there is no `M569 S1 T0 E` line in the output. This is the correct behavior.
* Execute `M500`.
* Power down the board, power it up again, execute `M503` and see that there still is no `M569 S1 T0 E` line.
* Power down the board, power it up just via USB, execute `M502` followed by `M500`.
* Power down the board, power it up via the main supply. Execute `M503` - there will be a `M569 S1 T0 E` line.

**Expected behavior:** [What you expect to happen]

`M502` followed by `M500` will initialize EEPROM to the values in `Configuration.h` regardless of whether the TMC drivers have full power.

**Actual behavior:** [What actually happens]

Executing `M502` followed by `M500` when board is only powered by USB results in different values in EEPROM than if full power is applied.

#### Additional Information

It looks like `MarlinSettings::save()` attempts to fetch the values from the drivers instead of using the cached values that are used by `M80` and `M122 I` to re-initialize the drivers. I noticed this specifically with StealthChop/Spreadcycle but code review shows that other TMC driver configuration values suffer from the same issue.

GMagician, 2020-09-09:
I think this will give you further issues if you don't "reset" tmc after full power. This was already seen, and reported, by me a "month ago". Some boards keep vmot to 3.3V. To fix this just give 5V to vmot when 12/24V is missing (if your board has possibility to do that).
Not sure trinamics report this, it can be a board design fault or missing info issue

ManuelMcLure, 2020-09-09:
This isn't about the initialization, but rather that even though Marlin knows what the values should be for the TMC drivers (in the case of the SpreadCycle/StealthChop these are in `stored.stealthChop_enabled`), instead of using those values when storing with `M500` it instead tries to read them from the TMC driver. If the values in EEPROM are correct when booting up (because they were stored with `M500` once full power was applied) they are correctly sent to the drivers when `M80` or `M122 I` is executed. It's just this save issue that is a problem. I'm going to do a PR to get as much of this stuff from the stored values instead of from the drivers.

ManuelMcLure, 2020-09-13:
Fixed by #19364

PR #19364 (2020-09-13): Use stored values for TMC StealthChop
### Description

Currently, Marlin attempts to read from the TMC drivers the values for whether StealthChop is enabled or not before saving them to EEPROM on an `M500`. If the TMC drivers are not powered on, this means that the common `M502`/`M500` operation to initialize EEPROM after a firmware update may result in the wrong values stored in EEPROM.

This appears to be the only TMC parameter that is read from the drivers on an `M500`. The hybrid threshold and current are saved from the stored cache values.

This PR uses the cached values for StealthChop i [... cut]
closed 2020-09-13 by ManuelMcLure