MarlinFirmware/Marlin #19315: [BUG] Compilation error, missing "libmaple/usart.h" but using arduino IDE

ReuvenStahl, 2020-09-09:
i'm trying to compile Bugfix for my MKS bored using the Arduino IDE.
and i'm having the same issue as :
https://github.com/MarlinFirmware/Marlin/issues/19305

any suggestions?

AnHardt, 2020-09-09:
> any suggestions?

Use VSCode.
Don't duplicate issues.
Follow the discussion in #19305 
Please close this issue.

closed 2020-09-09 by thisiskeithb

sjasonsmith, 2020-09-09:
@ReuvenStahl a patch is already posted. Try making the changes you find here:
https://github.com/MarlinFirmware/Marlin/pull/19307

That will probably fix your issue. I do agree with the recommendation to use PlatformIO. Nobody doing any serious development in Marlin still uses Arduino IDE, so it is easy for breaks like this to happen. Arduino IDE might seem simpler, but it is also much easier for things to go wrong.

sjasonsmith, 2020-09-09:
Duplicate of #19305