MarlinFirmware/Marlin #19316: [FR] Example configurations for Ender 3 Pro and SKR mini E3

piotrva, 2020-09-09:
### Description

It would be nice to have example configurations for Ender 3 Pro with SRK mini E3 boards (v1.0, v1.2, v2.0).

thisiskeithb, 2020-09-09:
The only difference is the default printer name and status screen graphic (if you enable it).

piotrva, 2020-09-09:
Thanks, I have just compared files. But would be nice to have it linked in folder for "Pro" for newbies.

thisiskeithb, 2020-09-10:
See https://github.com/MarlinFirmware/Configurations/pull/211.

closed 2020-09-10 by thinkyhead

piotrva, 2020-09-10:
Thanks - ultra fast response ;)

Deneteus, 2020-09-12:
Which one applies to the SKR E3-DIP V1.1?  :)

sjasonsmith, 2020-09-12:
@Deneteus I don’t think there is an example for the DIP version. It is probably most similar to the Mini E3 1.2. You will have to update the motherboard name and stepper driver types in your Configuration.h file.

Deneteus, 2020-09-12:
> @Deneteus I don’t think there is an example for the DIP version. It is probably most similar to the Mini E3 1.2. You will have to update the motherboard name and stepper driver types in your Configuration.h file.

I am aware. I am going to post this link here for reference just in case anyone asks.
https://www.makenprint.uk/3d-printing/3d-printing-guides/skr-e3-dip-marlin-2-setup-part-1/

Deneteus, 2020-09-12:
I found my answer. 
[image]