MarlinFirmware/Marlin #19320: [BUG] Tronxy XY-2 LVLchitu-v8-190729

Snow4DV, 2020-09-09:
# Bug Description

Hello! Trying to install Marlin with LVGL on Tronxy XY-2 (not pro) with Chitu V8 190729 board (in configuration V5 is set up as it was said in yt tutorial by Jeff). Everything works using the color or classic ui but not lvgl. Assets are on the sd card (dialog with preparing assets was shown on the first boot) but after that i am just getting a gray screen. If i try to click in some spot some other menues opened (Some screenshots: 
[image]
[image]
[image]

)

### My Configurations

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5196330/Configuration_adv.zip) 

Used custom one from Tronxy XY-2 pro/x5sa repo but modified it for XY2 

### Steps to Reproduce

Flash it on my hardware & boot it
#### Additional Information

There is another problem that can be connected with this one: 

If i enable LVGL and ADVANCED_PAUSE_FEATURE (for runout sensor) it doesnt let me compile it:

`In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:41:0,
                 from Marlin\src\HAL\STM32F1\HAL_SPI.cpp:35:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:825:6: error: #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
     #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:41:0,
                 from Marlin\src\HAL\STM32F1\Servo.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:825:6: error: #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
     #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:41:0,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:825:6: error: #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
     #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\dogm\../../../inc/MarlinConfig.h:41:0,
                 from Marlin\src\HAL\STM32F1\dogm\u8g_com_stm32duino_swspi.cpp:21:
Marlin\src\HAL\STM32F1\dogm\../../../inc/SanityCheck.h:825:6: error: #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
     #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:41:0,
                 from Marlin\src\HAL\STM32F1\eeprom_bl24cxx.cpp:28:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:825:6: error: #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
     #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:41:0,
                 from Marlin\src\HAL\STM32F1\eeprom_flash.cpp:32:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:825:6: error: #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
     #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
      ^~~~~
*** [.pio\build\chitu_v5_gpio_init\src\src\HAL\STM32F1\HAL_SPI.cpp.o] Error 1
*** [.pio\build\chitu_v5_gpio_init\src\src\HAL\STM32F1\Servo.cpp.o] Error 1
*** [.pio\build\chitu_v5_gpio_init\src\src\HAL\STM32F1\HAL.cpp.o] Error 1
*** [.pio\build\chitu_v5_gpio_init\src\src\HAL\STM32F1\dogm\u8g_com_stm32duino_swspi.cpp.o] Error 1
*** [.pio\build\chitu_v5_gpio_init\src\src\HAL\STM32F1\eeprom_bl24cxx.cpp.o] Error 1
*** [.pio\build\chitu_v5_gpio_init\src\src\HAL\STM32F1\eeprom_flash.cpp.o] Error 1`

However there are no problems with it on color ui.

Using the latest marlin firmware with my configuration

Snow4DV, 2020-09-09:
That's the repo: https://github.com/rhapsodyv/Marlin

UPD: Sorry, my mistake - used the wrong configuration_adv - that's why it didn't compile with ADVANCED_PAUSE_FEATURE

But even after replacing it with this one, the problem with blank screen is still on its place.

Correct configuration file: [Configuration_adv (2).zip](https://github.com/MarlinFirmware/Marlin/files/5196469/Configuration_adv.2.zip)

rhapsodyv, 2020-09-09:
Looks like your board have only 1mb, and not a 2mb like the x5sa.

Open the pins file and disable font in spi flash, and test again.
`Marlin/src/pins/stm32f1/pins_CHITU3D_V5.h`

Change `#define HAS_SPI_FLASH_FONT                   1` to `#define HAS_SPI_FLASH_FONT                  0`

Snow4DV, 2020-09-09:
> Looks like your board have only 1mb, and not a 2mb like the x5sa.
> 
> Open the pins file and disable font in spi flash, and test again.
> `Marlin/src/pins/stm32f1/pins_CHITU3D_V5.h`
> 
> Change `#define HAS_SPI_FLASH_FONT 1` to `#define HAS_SPI_FLASH_FONT 0`

After doing this it boots but screen doesn't turn on at all (even the backlight).

But it boots color ui with disabled HAS_SPI_FLASH_FONT without any problems

rhapsodyv, 2020-09-09:
> After doing this it boots but screen doesn't turn on at all (even the backlight).
>
> But it boots color ui with disabled HAS_SPI_FLASH_FONT without any problems

After you changed `HAS_SPI_FLASH_FONT` to 0, did you try to load the assets again?

Snow4DV, 2020-09-09:
> After you changed `HAS_SPI_FLASH_FONT` to 0, did you try to load the assets again?

Are you talking about deleting "_assets" folder? If yes i've just tried to and nothing changed

rhapsodyv, 2020-09-09:
Renaming it back to `assets`, while using the firmware with disabled font spi support...

Snow4DV, 2020-09-09:
> Renaming it back to `assets`, while using the firmware with disabled font spi support...

Oh. Ok, i tried to unpack the assets (two mks* folders and a bin file for robin) and also reverted to a firmware with spi, "initialized" the assets, renamed "_assets" to "assets" (just in case if these files after copying assets do not stay the same) - the same situation

Snow4DV, 2020-09-09:
> Renaming it back to `assets`, while using the firmware with disabled font spi support...

Ok, i just missed the fact that i got to unpack the assets' bins right to the assets folder :| 

Looks like everything works with the spi enabled

closed 2020-09-09 by Snow4DV

rhapsodyv, 2020-09-09:
You don't need the files from MKS too. Marlin will generate the `assets` folder in the build directory too, together with the `update.cdb` :-)

thisiskeithb, 2020-09-10:
These configs are not from Marlin's repo and we do not support forks here.