MarlinFirmware/Marlin #19321: [BUG] (sensorless homing with TMC2209 not working)

Mike8-I, 2020-09-09:
Description:

On my Prusa I3 clone which I changed from 8bit to 32bit with bigtreetech SKR1.4 with Bigtreetech TMC2209 v1.2 on Marlin 2.0.c from 06.09.2011 senssorless homing is not working .  Changing stall sensitivity just by one digit (eg. From 106 to 105) makes the difference from immediate stop oft he motor where ever the position is (106) to bumping into the frame and rattling (105).
Investigating the datasheet oft he TMC2209 shows that the stallguard4 of the TMC2209 is optimised for use with stealth chop (!) (All older TMCx are optimized for use with spread cycle). In advance.h it is stated that sensorless homing for TMCx is always done in spread cycle mode. My assumption is, that this ist he culprit with the TMC2209 sensorless homing not working.

from datasheet
11.1StallGuard4vs.StallGuard2
**StallGuard4 is optimized for operation with StealthChop, its predecessor StallGuard2 works with SpreadCycle**. The functionis similar: Both deliver a load value, going from a high value at lowload, to a low value at highload.While StallGuard2 becomes tuned to show a “0”-reading for stall detection, StallGuard4 uses a comparison-value to trigger stall detection, rather than shifting SG_RESULTitself

source [](https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2209_Datasheet_V103.pdf)

Mike8-I, 2020-09-09:
I meant Marlin 2.0.x bugfix

GMagician, 2020-09-09:
Not sure but TMC2209 code act differently. There has been a PR to fix it a long time ago

GMagician, 2020-09-09:
```c++
 bool tmc_enable_stallguard(TMC2209Stepper &st) {
    const bool stealthchop_was_enabled = !st.en_spreadCycle();

    st.TCOOLTHRS(0xFFFFF);
    st.en_spreadCycle(false);
    return stealthchop_was_enabled;
  }
```

that's the involved code

Mike8-I, 2020-09-09:
Then this comment:
* TMC2130, TMC2160, TMC2209, TMC2660, TMC5130, and TMC5160 only
   * Connect the stepper driver's DIAG1 pin to the X/Y endstop pin.
   * X, Y, and Z homing will always be done in spreadCycle mode.
Then this comment in advance.h is wrong....    and I have a different issue.......
Assume this can be closed then.

closed 2020-09-09 by thisiskeithb