MarlinFirmware/Marlin #19327: G2 | G3 have different feedrates

Protos58, 2020-09-10:
Marlin 2.0.5.3. CNC 3x.
G3 compared to G2 have different speeds. G2 have normal speed as indicated at F parametr.
However, G3 has a speed of 5 or 10 times less than G2.
Just ...
// G2/G3 Arc Support
#define ARC_SUPPORT

thinkyhead, 2020-09-10:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Protos58, 2020-09-11:
Thanks. In the process...

Protos58, 2020-09-11:
I have compiled this. And I got some features. Not about G2|G3.

- If I enable Spindle, I try to give up the regulation of the spindle power. However I have: "'SPINDLE_LASER_PWM_PIN' was not declared". And until I agree to put pwm control and determine the pin, the compilation does not work.
- Unable to release pin occupied by FUN0. I haven't FAN0 in my system. But I have #error "SPINDLE_LASER_PWM_PIN conflicts with FAN_PIN." (P2_03 on my board)

Protos58, 2020-09-11:
I did not try it on CNC. Maybe tomorrow.

Protos58, 2020-09-14:
Work fine. Thank You!

closed 2020-09-14 by Protos58