MarlinFirmware/Marlin #19328: Ender 5 Pro now shipping with v4.2.2 mainboard

dougbalish1, 2020-09-10:
Dont believe this has been addressed, if it has, please link me to where:

The default ender 5 pro config now no longer completely works for (what i assume to be) most ender 5's shipping. I, and a few people ive been in contact with, have received ender 5's with creality v4.2.2 mainboards, and while creality has a .bin that works with both the stock version and bltouch (i have a bltouch) version on their site, they refuse to share the (marlin v2 based) source. 

I would think at least a warning in the config for ender 5 pros is in order to prevent wasted time, as well as possibly an ender 5 pro v1.5 type addition to the configs. 

I would love to help contribute, but i am new to github and open source (competent enough c programmer/debugger though) so if anyone wants to point me in the right direction to help, i would be glad. 

Some additional info in case it helps anyone: I have tried building the ender 5 pro config but using the STM32F103RET6//Creality board V4 settings transplanted into the ender 5 pro config, and i can get this to work with some tweaks, but not the bltouch. This indicates (to me at least) that the pinout is different. If anyone would like me to help debug also, i have access to a cheap oscope and multimeter, fpga, logic analyzer, etc.

SlackHacky, 2020-09-10:
because there are to date 3 ways to connect a BLTouch on to a v4.2.2 (or v4.2.7) board 
1) All five pins into the Bltouch port
2) 3 pins in BlTouch port and 2 in Z stop port
3)Creality's err way (to use up old v1/v2 SKU kits) is 2 pins into z stop port and connect a adaptor board into the screen display port...

so its all about which port you pick and how you define the configuration.h file

dougbalish1, 2020-09-10:
> because there are to date 3 ways to connect a BLTouch on to a v4.2.2 (or v4.2.7) board
>
> 1. All five pins into the Bltouch port
> 2. 3 pins in BlTouch port and 2 in Z stop port
> 3. Creality's err way [...]

Thanks for the reply, i was wondering what the Bl_T header on the mainboard was for since they shipped me the adapter board, makes sense now though. 

To clarify: What you are saying is that marlin currently defines the bltouch to be connected through option 1 or 2 (but option 1 is the default that would not require any pin redef?) for 4.2.2 mainboards, but since i currently have option 3, which is not the officially supported way, my best bet is to rewire it to the bltouch port/zstop, because otherwise i would have to trace the pin for signal and the z endstop and redefine bltouch_probe as signal pin, and bltouch_in/out as the endstop pins (either in pins.h or override it in the config)?

SlackHacky, 2020-09-16:
I use option 1 when I self compile, this is not a BUG as such, more no clear documentation for a User that did not know these configurations.

sjasonsmith, 2020-11-12:
@thisiskeithb what do you think about this one? Has this been addressed? Do we need to tweak examples to help with this?