MarlinFirmware/Marlin #19329: Help me Arduino does not build marlin 2.0.6.1 firmware

Subia75, 2020-09-10:
Hi everyone
i am trying to compile the firmware for my anycubic I3 mega S with marlin 2.0.6.1 before i had version 2.0.5 but i had not compiled it myself.

When I go to modify the configuration.h file to use my TFT35 display, arduino ide is no longer able to build the firmware and gives me an error.
The U8glib library are installed so I don't understand what that could be.

This is the Arduino repport:
"
  1.9.0-beta (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

fork / exec C: \ Users \ alessio.tius \ AppData \ Local \ Arduino15 \ packages \ arduino \ tools \ avr-gcc \ 7.3.0-atmel3.6.1-arduino7 / bin / avr-gcc.exe: Name of the file or extension too long.
Error compiling for board Arduino Mega or Mega 2560.
"

I am attaching a screenshot of the changes made to the configuration.h file compared to the pre-compiled one found on github.com.
[image]

I ask if anyone can help me.

thinkyhead, 2020-09-10:
Try moving your project folder closer to the root of your hard drive. Or, use PlatformIO which builds the firmware much faster and produces a more optimized binary.

Subia75, 2020-09-10:
Thanks for adjust my topic.
I tried with VScode to build and it works.
With arduino it also fails by moving the folder to the root.
I get the error again when I try to read the file "avr-gcc.exe" in the folder path I posted at the beginning.
The problem seems to be due to the display setting using the "avr-gcc.exe" file.
Anyway I will use VScode in the future which is more comfortable for me to modify.

ellensp, 2020-09-10:
This is a well documented limitation of Windows.
It has also been mentioned here about 100 time here...
In windows the Fork system call can take a maximum path and executable name of 32k characters.
The way Arduino IDE tries to link marlin together at the end of the compile is to joins all the .o files with one very long single command that exceeds this limit.

The easiest way to avoid this is to use vscode with platformio. 
The second best way is to avoid Windows.
The third is to manually remove unused directories from marlin that don't actually need to be compiled for you system (platformio automatically does this) 
What directories is system specific and you need to have in depth marlin knowledge to know what is safe to remove.
Eg for a AVR based controller like you have you can remove the following from Marlin/src/HAL You don't need directories ESP32, LINUX, STM32F1, TEENSY31_32, DUE, LPC1768, SAMD51, STM32, STM32_F4_F7, TEENSY35_36.

closed 2020-09-10 by ellensp