MarlinFirmware/Marlin #19334: [BUG] Cannot upload Marlin2.0.x (delta) to my MKS SBASE

keqinn, 2020-09-10:
### Bug Description

Hello everybody!

I have downloaded a lot of Marlin2.0 firmwares derectly from Marlin webpage, such as 2.0.5.3, 2.0.5.4, 2.0.6.1, and I successfully compiled them with VScode and PlatformIO. 

However, I cannot upload all of these firmwares to my MKS SBASE successfully.

After I copy firmware.bin into SD card, I cannot use my software( Repetier Host) to control my printer. Repetier tells me that there are some commands waiting for sending when I connect my computer and printer. And the LED on my MKS SBASE does not blink anymore( I successfully upload Marlin2.0.0.1 before, and the LED will blink).

I hope that I could upload a new version of Marlin on my NKS SBASE.

Many thanks for any help.

### My Configurations

I get my configuration.h and configuration_adv.h derectly from the config/delta/MKS/SBASE folder.
And I just change "#define TEMP_SENSOR_0 -1" to "#define TEMP_SENSOR_0 1".

### Steps to Reproduce

1. Download Marlin firmware from github, such as 2.0.6.1 or 2.0.5.4.
2. Change configuration.h and configuration_adv.h.
3. Complie with VScode and PlatformIO. (Successfu)
4. Upload firmware.bin to MKS SBASE.(fail)

**Expected behavior:

I can control my 3d printer with Repetier Host, and the led on MKS SBASE will blink.

**Actual behavior:** [What actually happens]

Repetier tells me that there are some commands waiting for sending when I connect my computer and printer.
 And the LED on my MKS SBASE does not blink anymore.

ellensp, 2020-09-10:
The example from https://github.com/MarlinFirmware/Configurations/blob/import-2.0.x/config/examples/delta/MKS/SBASE/Configuration.h  seems to be wrong.

It contains 
```
#define SERIAL_PORT 0

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
//#define SERIAL_PORT_2 -1
```
For USB communication you need one of these two serial ports to be -1

Serial port 0 is probably for a TFT or Wifi modules if you dont have these. Change #define SERIAL_PORT 0 to #define SERIAL_PORT -1

If you do have these modules simply enable #define SERIAL_PORT_2 -1  (by remove preceding //)

NB

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-09-10 by ellensp

sjasonsmith, 2020-09-10:
This is interesting, because I myself was using an sBase configured for port 0 for a while, and it worked. Something changed at one point and I was forced to change it to -1 about a year ago. We honestly don’t know why it was working with 0 ever, but it certainly did.

keqinn, 2020-09-11:
> The example from https://github.com/MarlinFirmware/Configurations/blob/import-2.0.x/config/examples/delta/MKS/SBASE/Configuration.h seems to be wrong.
>
> It contains
>
> ```
> #define SERIAL_PORT 0
>
> /**
>  * [...]

Wonderful solution! Now I can control my 3D printer again, thanks for your help! ^_^

keqinn, 2020-09-11:
@ellensp Wonderful solution, now I can control my 3D printer again! Thanks for your sharing! ^_^

keqinn, 2020-09-11:
@sjasonsmith Yeah, I have the same experience with you! Thanks for your info, it does help to me.