MarlinFirmware/Marlin #19338: [FR] Support Segger jLink debugger

mron, 2020-09-10:
#19088 # Description
Add SWD debugger support for ARM based processors
### Feature Workflow
Add a debug tool support in platformio

#### Additional Information
I have no idea how to do this. I've used it with Zephyr. It's great

sjasonsmith, 2020-09-10:
This is already supported by PlatformIO, and can be used with Marlin. The actual configuration needed differs somewhat for every board and MCU. I have myself used a jLink with an LPC1768 board, although I typically have much greater success live debugging with STM32 environments (and an ST-Link).

This is an advanced use-case that I wouldn't encourage people to use, unless they really know what they are doing. If people really know what they are doing they can add the configuration options as needed, and post pull requests to add some improved defaults to Marlin.

For now I am going to close this feature request. I really don't see any chance that someone is going to proactively act upon this,  outside of whatever particular board they are currently working with.

closed 2020-09-10 by sjasonsmith

Bob-the-Kuhn, 2020-09-11:
I personally like the J-Link when used with Segger's Ozone package.