MarlinFirmware/Marlin #19342: [BUG] SKR Pro v1.2 Linux USB serial printing broken

Yenya, 2020-09-11:
### Bug Description

Host software on Linux cannot talk to SKR Pro v1.2 over the USB serial connection. The serial port is visible as `/dev/ttyACMx`, but connection fails with different symptoms for each host software I tried (RepetierHost, Slic3r, Pronterface).

### My Configurations

[skrpro-config.zip](https://github.com/MarlinFirmware/Marlin/files/5207416/skrpro-config.zip)

### Steps to Reproduce

1. install the firmware with `#define SERIAL_PORT -1` (USB-emulated serial port).
2. Power up the SKR Pro v1.2 and connect it over USB cable to a Linux computer. Make sure that the two USB jumpers J49 near the USB port are set to the "USB" side, not the "Host" side.
3. The serial device `/dev/ttyACMx` should appear in the Linux box.
4. Verify that you have access to that device (run `chmod a+rw /dev/ttyACM1` as root and disable SELinux, if needed).
5. Start up the host software (RepetierHost, Slic3r, Pronterface), and try connecting to that device.

**Expected behavior:**

The connection should be established, and the printer should report for example bed and nozzle temperatures.

**Actual behavior:**

Slic3r: it tries to connect for about 5 seconds, and the fails with "`Connection failed. Check serial port and speed`." error message. According to `strace`(1), it sends out the M105 command, but does not get anything back.

RepetierHost: It connects (the connection button gets green), but immediately reports "`7 commands waiting`" in the status line, and these commands stay in the queue without response forever.

Pronterface: connection fails and the following message appears in the kernel log (dmesg): `cdc_acm 1-10:1.0: failed to set dtr/rts`

#### Additional Information

I am able to connect to `/dev/ttyACM1` using `minicom`, send the commands, and get responses most of the time. Sometimes the output is garbled, though. For example, when I open a terminal with "`cat /dev/ttyACM1`" command, and run "`echo M105 >/dev/ttyACM1`" in another terminal repeatedly, I got the following responses back in the first terminal:

```
ok T:-15.00 /0.00 B:-15.00 /0.00 P:-15.00 /0.00 @:0 B@:0
echo:Unknown command: "ok T:-15.00 /0.00 B:-15.00 /0.00 P:-15.00 /0.00 @:0 B@:0"
ok
echo:Unknown command: "echo:Unknown comma0 B:-15.00 /0.00M105"
ok
echo:Unknown command: "echo:Unknown commM105"
ok
echo:Unknown command: "echo:Unknown commaM105"
ok
echo:Unknown command: "echo:Unknown commM105"
ok
echo:Unknown command: "echo:Unknown commokM105"
ok
echo:Unknown command: "echo:Unknown commM105"
ok
echo:Unknown command: "echo:Unknown commM105"
ok
echo:Unknown command: "echo:Unknown comhoM105"
ok
echo:Unknown command: "echo:Unknown commM105"
ok
echo:Unknown command: "echo:Unknown commM105"
ok
echo:Unknown command: "echo:Unknown commM105"
ok
echo:Unknown command: "echo:Unknown commM105"
ok
echo:Unknown command: "echo:Unknown comma commM105""
ok
echo:Unknown command: "ok"
ok
ok T:-15.00 /0.00 B:-15.00 /0.00 P:-15.00 /0.00 @:0 B@:0
echo:Unknown command: "echo:Unknown omchok B:-15.00 /0.00 P:-15.00 /0.00 @:0 B@:0"
ok
```

I managed to get slic3r to connect by running "`echo M105 >/dev/ttyACM1`" from another terminal during these 5 seconds slic3r waits for the initial response from the printer.

I am sometimes able to connect with Pronterface, but not reliably.

FWIW, I also have BTT-TFT335-E3 connected over serial port 1 (`#define SERIAL_PORT_2 1`). This seems to communicate reliably with SKR Pro board.

The host software itself (slic3r, RepetierHost, and Pronterface) works correctly on the same computer with my Arduino Mega + RAMPS 1.4 + Marlin 1.9 setup.

I tested the both the current `2.0.x` branch (9913333c33d234264a9072630e8b5ffbf07bad7e), and the `bugfix-2.0.x` branch (4268d86da18ce8ffffc8236e0895fd2f3ba7f95d).

I reported it also to the SKR Pro repository, but I am not sure whether it is the right place, so I am reporting this also here. Maybe the problem is in the way how the SKR Pro MCU emulates the USB serial port.

https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/165

Thanks for any hints what else should I try.

sjasonsmith, 2020-09-11:
Do you another system you can use to test the USB-Serial connection? Perhaps a Windows computer or a Raspberry Pi?

Yenya, 2020-09-12:
Yes, I have tested this on my workstation and on my laptop. Both running Fedora Linux, though.

lamikr, 2020-09-17:
Could it be possible that some other software like modemmanager try to communicate automatically with the /dev/ttyACM0 after your system detects it and that causes some problem? I have had that type of problems with some other devices some years ago and the fix was to disable at least the modem manager. Couple of links for that type of problems

https://unix.stackexchange.com/questions/270989/disable-automatic-serial-port-probing-in-linuxfedora
https://github.com/tinyfpga/TinyFPGA-BX/issues/4

Yenya, 2020-09-21:
Update:

* the binary firmware which was distributed with the SKR Pro board itself works with Slic3r, RepetierHost, and Pronterface.
* I have tried SKR v1.4 Turbo board with the same symptoms: Slic3r cannot connect after 5 s delay, RepetierHost connects but immediately reports 7 commands waiting, and Pronterface works correctly.

Yenya, 2020-10-22:
The issue is still present (also in SKR v1.4 Turbo as described above). Tested also Marlin 2.0.7.2.

ellensp, 2020-10-22:
If its working with one host program, ie pronterface, the issue must be with the other host programs.

Yenya, 2020-10-22:
I would be _very_ careful with this "must be". It can merely be that the other programs use the serial port differently (e.g. different order of TTY ioctl() calls, or something like that). After all, with Arduino Mega + RAMPS all three programs work.

Maybe something in the ARM HAL for USB serial port emulation?

ellensp, 2020-10-22:
To me it just looka like the host software needs updated, to understand modern marlin response strings. Like pronterface has been regularly.

Yenya, 2020-10-22:
Does the cat(1) command also need to be updated, as in my the "Additional information" section of my report above? :-)

ellensp, 2020-10-22:
your fancy redirection test test above is clearly breaking communications "echo:Unknown comma0 B:-15.00 /0.00M105" is not a valid gcode. part of the reply is going back as input..   you have some sort of loop
see https://3dprinting.stackexchange.com/questions/5382/issues-with-direct-g-code-transmission-via-serial-port for some pointers.

Vertabreak, 2020-10-22:
just wana add the latest version of slic4r is dated back in 2018 https://github.com/slic3r/Slic3r
repetier lists the latest version being for ubuntu zesty dating it to around 2017 https://www.repetier.com/
marlin 2.0 being released in 2019 it would seem to me that it would be very possible that slic3r and repetier have become so old they no longer function with the latest marlin.

Yenya, 2020-10-22:
Slic3r is being actively developed, they just didn't make a release recently. And indeed, I tested this also against the current master with the same results.

RepetierHost is going proprietary, so you are right that it is not the main problem of this issue. I just wanted to test it to verify that it is not a problem of slic3r (or cat(1) :-).

sjasonsmith, 2020-11-07:
Is there anything actionable here? I have seen lots of discussion about gcode sender tools and their issues. Is there any possibility of a Marlin bug here?

ellensp, 2020-11-07:
I don't believe so. Just misunderstandings and bad test procedures.

sjasonsmith, 2020-11-07:
Ok, I will close this as it doesn’t seem there is anything actionable in the side of Marlin. Please reply if you feel this was incorrect, and explain why.

closed 2020-11-07 by sjasonsmith