MarlinFirmware/Marlin #19343: [BUG] RAM overflow with TFT 480x320 SPI

tomthemod, 2020-09-11:
### Bug Description

When compiling Marlin last bugfix with some configured options for an BTT SKR 1.4 (LPC1768 = 32Kb RAM) :
- with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER lcd, RAM usage : 14304 bytes (18432 bytes free)
- with TFT_480x320_SPI : compiler says : "region `RAM' overflowed by 29032 bytes"

### My Configurations

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/5208501/Configurations.zip)

### Steps to Reproduce

1. In my configuation.h, comment L2226 and uncomment L2003, and compile in vscode/platformio : see approx. 14304 bytes of ram used in lp
2. take my initial configuration.h, or 1. uncomment L2226 and comment L2003, and compile in vscode/platformio : you will see a big ram overflow.

#### Additional Information

With discount graphic card :
Checking size .pio\build\LPC1768\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  43.7% (used 14304 bytes from 32736 bytes)
Flash: [====      ]  38.7% (used 183876 bytes from 475136 bytes)

With TFT SPI 
./.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio\build\LPC1768\firmware.elf section `.bss' will not fit in region `RAM'
./.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: region RAM overflowed with stack
./.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: region `RAM' overflowed by 29032 bytes

ellensp, 2020-09-11:
This is a limitation of your controller. Its not a bug!

closed 2020-09-11 by ellensp

rhapsodyv, 2020-09-11:
Just set TFT_BUFFER_SIZE for a smaller value... 
for example:
`#define TFT_BUFFER_SIZE 2000`