MarlinFirmware/Marlin #19346: [BUG] (short description)

ase95, 2020-09-11:
### Bug Description

I've edited the marlin 2.0 firmware on VS for my tevo tornado but for some reason when i upload it i get stuck on the boot screen,it keeps reloading the custom boot and the marlin.
my board is a BOARD_MKS_GEN_L

### My Configurations

[My config](https://github.com/MarlinFirmware/Marlin/files/5210094/marlin20.zip)

### Steps to Reproduce
just install the firmware and the issue should present itself rightaway

**Expected behavior:** 
Normal  behavior of the printer

**Actual behavior:** 
stuck in the booting screen with custom screen and marlin that keeps restarting

nvumnov, 2020-09-11:
Здравствуйте, после обновления Marlin, не показывает экран.
mks SGEN L + Fysetc Mini 12864 V2.1
Вот мои файлы: 
[SGEN L + Fysetc Mini 12864 V2.1.zip](https://github.com/MarlinFirmware/Marlin/files/5210442/SGEN.L.%2B.Fysetc.Mini.12864.V2.1.zip)
Помогите.

ase95, 2020-09-11:
> might be worth changing the title so its descripe
> 
> [BUG] (short description)
> 
> does not tell much
> 
> did you try
> [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) ?

yikes, i completly forgot about changing the title ahah.
i've not tried that , i'll do it asap and  i'll let you know the result.
do you think i can just put my config in there or should i re do it from scratch?

ase95, 2020-09-11:
> Здравствуйте, после обновления Marlin, не показывает экран.
> mks SGEN L + Fysetc Mini 12864 V2.1
> Вот мои файлы:
> [SGEN L + Fysetc Mini 12864 V2.1.zip](https://github.com/MarlinFirmware/Marlin/files/5210442/SGEN.L.%2B.Fysetc.Mini.12864.V2.1.zip)
> Помогите.

i don't speak russian... could you write it in english?

nvumnov, 2020-09-11:
Hello, after Marlin's update, doesn't show the screen.
mks SGEN L + Fysetc Mini 12864 V2.1
Here are my files:
[SGEN L + Fysetc Mini 12864 V2.1.zip](https://github.com/MarlinFirmware/Marlin/files/5210474/SGEN.L.%2B.Fysetc.Mini.12864.V2.1.zip)
HELP!!!

rhapsodyv, 2020-09-11:
@thisiskeithb is this related with that other LCD we were working?

thisiskeithb, 2020-09-11:
> is this related with that other LCD we were working?

I was using an MKS Mini 12864, not the Fysetc Mini.

sjasonsmith, 2020-09-11:
There are actually two completely unique issues reported here now.

The original report is an MKS_GEN_L with a RepRapDiscount Full Graphics.

sjasonsmith, 2020-09-11:
I have the hardware for both of these...I guess I'll assign it to myself so I remember to try them.

rhapsodyv, 2020-09-11:
@thisiskeithb could that that contrast setup what we removed, affect fysetc mini? I will take a look if the share the same code.

We removed that `#if`
```
void MarlinUI::init_lcd() {
   #if DISABLED(MKS_LCD12864)
```

sjasonsmith, 2020-09-12:
@ase95 your configurations boot just fine for me, on an MKS Gen-L V1.0 with a BigTreeTech RepRapDiscount Full Graphic Smart Controller. I changed nothing, and used them on top of the current bugfix-2.0.x branch, commit 6f8282f01ebdb0f5594b84a1ff2462a8aba10961.

sjasonsmith, 2020-09-12:
@nvumnov I agree that something seems wrong with the board and LCD you are using. I can't get them to work either.

Please open a new issue for your problem. It isn't related to the original issue, since it is completely different hardware.

ase95, 2020-09-12:
> @ase95 your configurations boot just fine for me, on an MKS Gen-L V1.0 with a BigTreeTech RepRapDiscount Full Graphic Smart Controller. I changed nothing, and used them on top of the current [...]

i've downloaded the bugfix2.0 and just used the same config but problem still occur 
[video](https://github.com/MarlinFirmware/Marlin/files/5212417/video.zip)

ase95, 2020-09-12:
> @ase95 your configurations boot just fine for me, on an MKS Gen-L V1.0 with a BigTreeTech RepRapDiscount Full Graphic Smart Controller. I changed nothing, and used them on top of the current [...]

i tried redoing the config file but same problem occurs... i can't understand what im doing wrong...

ase95, 2020-09-14:
> @ase95 your configurations boot just fine for me, on an MKS Gen-L V1.0 with a BigTreeTech RepRapDiscount Full Graphic Smart Controller. I changed nothing, and used them on top of the current [...]

Any idea on why is it not working for me ? i even tried using Atom instead of VS, marlin 2.0 and 2.0bugfix

ellensp, 2020-10-05:
@ase95 as a test (do not run like this) 
disable #define USE_WATCHDOG in Configuration.h 
See if anything changes

boelle, 2020-10-13:
@ase95 did you try what @ellensp suggested?

ase95, 2020-10-15:
i've not tried it. for now i'll settle for the 1.9 version and i'll wait for new update of the 2x to see if it get fixed

nongcj, 2020-10-18:
I downloaded the latest official version 2.0.7.2 today. The problem is the same as @ase95. When I turn on the switch, Marlin restarts repeatedly. My board is MKS_SGEN_L, and the display is MKS 12864A+3DTOUCH. I found that 3D TOUCH conflicts with MKS12864A. I disable BLTOUCH Marlin to start normally, or disable #define MKS_LCD12864, and connect PC with Pronterface to control the printer normally. I tried to disable @ellensp suggested #define USE_WATCHDOG with the same result. This problem has always existed from 2.0.6.1 to 2.0.7.2. I hope the Marlin team can solve this problem. Now I can only use 2.0.5.4 to wait for the latest update, thanks @boelle

thisiskeithb, 2020-10-18:
> My board is MKS_SGEN_L, and the display is MKS 12864A

@nongcj This issue is for the Gen-L + RepRapDiscount Full Graphic LCD and is unrelated to your hardware combo.

MKS has documented the A variant to only be compatible with 8bit/AVR boards. You’d need to use the B version on 32bit.

thisiskeithb, 2020-10-18:
> i've not tried it. for now i'll settle for the 1.9 version and i'll wait for new update of the 2x to see if it get fixed

@ase95: `bugfix-2.0.x` is updated daily, but if you don’t want to try what has been suggested here, I’ll close the issue for now.

closed 2020-10-18 by thisiskeithb