MarlinFirmware/Marlin #19350: [BUG] cr10s screen overlapped

Liger0, 2020-09-11:
### Bug Description

The text overlap the values on 2.0.6.1

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5211735/Marlin.zip)

### Steps to Reproduce

1. Go in a configuration

**Expected behavior:** Text are separate from the values

**Actual behavior:** Text hides the values

#### Additional Information

[image]

boelle, 2020-09-11:
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

Liger0, 2020-09-11:
Do you mean to test on latest bugfix or to test the stable releases?

ellensp, 2020-09-12:
Checked on current bugfix.

ellensp, 2020-09-12:
Something to do with how the wildcard in MSG_EN_STEPS = _UxGT("* Steps/mm"); is interpreted.
instead of resulting in "E1 Steps/mm"  Its resulting in "E1 Steps/mm Steps/mm" 
You cant see it on the steps/mm screen but you see it when you edit the value..

ellensp, 2020-09-12:
removing line 60 from https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/lcd/lcdprint.cpp#L60 
Fixes it. Trying to work out what this line was meant to do and if removing it causes any further issues.

ellensp, 2020-09-12:
Ok line 60 prints while checking the text isn't larger than the lcd. vs default which doesn't..
So adding a break is a better option.

ellensp, 2020-09-12:
Check the PR above works for you also.

ellensp, 2020-09-13:
This has been fixed in current bugfix

closed 2020-09-13 by ellensp