MarlinFirmware/Marlin #19352: [BUG]  SD-card not working with BOARD_ULTRATRONICS_PRO

annhan, 2020-09-12:
### Bug Description

 - I can't init my sdcard in Marlin with BOARD_ULTRATRONICS_PRO Board.
 - It's always `echo:No SD card` state when the SD card was inserted

### My Configurations

- branch : bugfix-2.0.x

ellensp, 2020-09-12:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

annhan, 2020-09-19:
It's an issue with Pin in ultratronic board.
Add
```
  #define SCK_PIN           76
  #define MISO_PIN          74
  #define MOSI_PIN          75
```
in pins_ULTRATRONICS_PRO.h for fix this issue

ellensp, 2020-09-19:
Reopeing, this does need added to the pins file

ellensp, 2020-09-19:
These have been added into marlin bugfix and will be in future releases

closed 2020-09-19 by ellensp