MarlinFirmware/Marlin #19353: [BUG] Power Loss Recovery freeze lcd display

ils15, 2020-09-12:
When i activate power loss recovey with MKS ROBIN NANO V1.2 and color UI, display freezes, sometimes when i start a new print, sometimes imediatelly

you can see my marlin code here
https://github.com/ils15/BlueR

boelle, 2020-09-12:
is your marlin a copy of [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) ?

ils15, 2020-09-12:
Marlin 2.0.6.1

boelle, 2020-09-12:
try with [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) instead, it's being updated daily

2.0.6.1 is just a frozen copy of [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x)

andreibobirica, 2020-09-13:
I have tried with the last bugfix and the problem is here too

ils15, 2020-09-13:
> try with [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) instead, it's being updated daily
> 
> 2.0.6.1 is just a frozen copy of [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x)

Same error

closed 2020-10-21 by github-actions[bot]