MarlinFirmware/Marlin #19354: [BUG] SPI Redefinition errors

DexterHill0, 2020-09-12:
### Bug Description

In the SPI.cpp file there are a few redefinition errors being thrown as well as two others errors. This is the second compilation of Marlin I have done. The first was successful. I then went on to enable NEOPIXEL leds and I started to get some error. I reverted back to a version without NEOPIXEL but I am continuing to get the errors.

### My Configurations

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/5212578/configs.zip)

### Steps to Reproduce

N/A

#### Additional Information
I am compiling for the SKR MINI E3 V1.2

Since I was having many issues before, I totally reinstalled VSCode and platformio. I then also used some the [pre-made configs](https://github.com/MarlinFirmware/Marlin/tree/2.0.x/config) and then made more changes to them. This time I also installed Marlin Auto Build.
The full log is here:
<details><summary>Log</summary>
<p>

```
platformio run -e STM32F103RC_btt_512K ; echo "done" >/var/folders/nv/3rvpqg7n2h37snbb2cf6pf280000gq/T/ipc

The default interactive shell is now zsh.
To update your account to use zsh, please run `chsh -s /bin/zsh`.
For more details, please visit https://support.apple.com/kb/HT208050.
iMac-269:Marlin-2.0.x 2 everyone$ platformio run -e STM32F103RC_btt_512K ; echo "done" >/var/folders/nv/3rvpqg7n2h37snbb2cf6pf280000gq/T/ipc
***************************************************************************************************************************************************************************************************************************************************
Obsolete PIO Core v5.0.1a1 is used (previous was 5.0.1)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-platformio-cores-in-a-system
***************************************************************************************************************************************************************************************************************************************************
Processing STM32F103RC_btt_512K (platform: ststm32@~6.1.0; board: genericSTM32F103RC; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 (6.1.1) > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32-maple 2.10000.200103 (1.0.0) 
 - tool-stm32duino 1.0.1 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareSerialM> 1.0.0
|-- <TMCStepper> 0.7.1
|   |-- <SPI> 1.0
|   |-- <SoftwareSerialM> 1.0.0
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <STM32ADC> 1.0
|-- <SPI> 1.0
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.91
|-- <Wire> 1.0
|-- <FreeRTOS701>
|-- <Servo(STM32F1)> 1.1.2
Building in release mode
Compiling .pio/build/STM32F103RC_btt_512K/FrameworkArduinoVariant/board.cpp.o
Compiling .pio/build/STM32F103RC_btt_512K/FrameworkArduinoVariant/wirish/boards.cpp.o
Compiling .pio/build/STM32F103RC_btt_512K/FrameworkArduinoVariant/wirish/boards_setup.cpp.o
Compiling .pio/build/STM32F103RC_btt_512K/FrameworkArduinoVariant/wirish/start.S.o
Compiling .pio/build/STM32F103RC_btt_512K/FrameworkArduinoVariant/wirish/start_c.c.o
Compiling .pio/build/STM32F103RC_btt_512K/FrameworkArduinoVariant/wirish/syscalls.c.o
[... 95 lines cut ...]
 void SPIClass::beginTransaction(uint8_t pin, const SPISettings &settings) {
      ^~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio/build/STM32F103RC_btt_512K/src/src/HAL/STM32F1/SPI.cpp.o] Error 1
In file included from Marlin/src/HAL/STM32F1/HAL_SPI.cpp:36:0:
/Users/everyone/.platformio/packages/framework-arduinoststm32-maple/STM32F1/libraries/SPI/src/SPI.h:171:15: warning: '_spi3_this' defined but not used [-Wunused-variable]
 static void (*_spi3_this);
               ^~~~~~~~~~
/Users/everyone/.platformio/packages/framework-arduinoststm32-maple/STM32F1/libraries/SPI/src/SPI.h:168:15: warning: '_spi2_this' defined but not used [-Wunused-variable]
 static void (*_spi2_this);
               ^~~~~~~~~~
/Users/everyone/.platformio/packages/framework-arduinoststm32-maple/STM32F1/libraries/SPI/src/SPI.h:165:15: warning: '_spi1_this' defined but not used [-Wunused-variable]
 static void (*_spi1_this);
               ^~~~~~~~~~
=========================================================================================================== [FAILED] Took 9.64 seconds ===========================================================================================================

Environment           Status    Duration
--------------------  --------  ------------
STM32F103RC_btt_512K  FAILED    00:00:09.636
====================================================================================================== 1 failed, 0 succeeded in 00:00:09.636 ======================================================================================================
iMac-269:Marlin-2.0.x 2 everyone$ 
```

</p>
</details>

ellensp, 2020-09-12:
you have older code. This doesn't happen in current bugfix.

In Platfmoio.ini you need the following (in particular "lib_ignore    = SPI"
```
[common_stm32f1]
platform      = ststm32@~6.1
build_flags   = !python Marlin/src/HAL/STM32F1/build_flags.py
  ${common.build_flags} -std=gnu++14 -DHAVE_SW_SERIAL
build_unflags = -std=gnu11 -std=gnu++11
src_filter    = ${common.default_src_filter} +<src/HAL/STM32F1>
lib_ignore    = SPI
lib_deps      = ${common.lib_deps}
  SoftwareSerialM
```

closed 2020-09-12 by ellensp

DexterHill0, 2020-09-12:
To my surprise that hasn't fixed it. The lib ignore is already in the platformio file:
```
#
# HAL/STM32F1 Common Environment values
#
[common_stm32f1]
platform      = ${common_stm32.platform}
build_flags   = !python Marlin/src/HAL/STM32F1/build_flags.py
  ${common.build_flags} -std=gnu++14 -DHAVE_SW_SERIAL
build_unflags = -std=gnu11 -std=gnu++11
src_filter    = ${common.default_src_filter} +<src/HAL/STM32F1>
lib_ignore    = SPI
lib_deps      = ${common.lib_deps}
  SoftwareSerialM
```
This is unedited file (also the one in the zip file). I also tried adding it to `[common_stm32]` but that hasn't fixed it. The only reason I am using outdated firmware is because (at least before reinstalling everything) it just didn't fully compile properly. The version now is the only one that has successfully compiled.

DexterHill0, 2020-09-12:
To my surprise that hasn't fixed it. The lib ignore is already in the platformio file:
```
#
# HAL/STM32F1 Common Environment values
#
[common_stm32f1]
platform      = ${common_stm32.platform}
build_flags   = !python Marlin/src/HAL/STM32F1/build_flags.py
  ${common.build_flags} -std=gnu++14 -DHAVE_SW_SERIAL
build_unflags = -std=gnu11 -std=gnu++11
src_filter    = ${common.default_src_filter} +<src/HAL/STM32F1>
lib_ignore    = SPI
lib_deps      = ${common.lib_deps}
  SoftwareSerialM
```
This is the unedited file (the one in the zip). I also tried adding it to [common_stm32] but that hasn't fixed it. The only reason I am using outdated firmware is because (at least before reinstalling everything) it just didn't fully compile properly. The version now is the only one that has successfully compiled.

ellensp, 2020-09-12:
you have a  "lib_ignore        =" in your [env:STM32F103RC_btt_512K] 
This removes the previously defined libs to ignore.

ellensp, 2020-09-12:
Neopixels on that processor is extremely hit and miss.. Thus is was disabled with warning to not expect it to work.
but ...

I have had others who have gotten it to work with this library
in platformio.ini find line  NEOPIXEL_LED  = Adafruit NeoPixel@1.5.0
change to NEOPIXEL_LED  = Adafruit NeoPixel=https://github.com/ellensp/Adafruit_NeoPixel
But there are no guarantees!

ellensp, 2020-09-12:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

DexterHill0, 2020-09-13:
Removing the other lib_ignore has fixed it, so thanks you!