MarlinFirmware/Marlin #19356: [BUG] Kingroon KP3 - LCD Touchscreen freezes if printer turned on without SD card (error msg SD Init Failed)

Hokeric, 2020-09-12:
[Marlin-Config Files.zip](https://github.com/MarlinFirmware/Marlin/files/5212649/Marlin-Config.Files.zip)

### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step] Take out SD card from printer
2. [Second Step] Turn on printer (without SD card)
3. [and so on...]

**Expected behavior:** [What you expect to happen]
LCD touch screen will respond to touch controls
There should not be "SD init failed" error message (as in 2.0.6)

**Actual behavior:** [What actually happens]
LCD does not respond to touch control
LCD shows "SD init failed" when printer is turned on

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2020-09-12:
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

Hokeric, 2020-09-13:
Have just tested the bugfix branch given. The "SD Init Failed" message was gone, and the touchscreen can be operated when the printer is turned on without SD card.

However, SD card removal and insertion are not automatically detected like in version 2.0.6. When the SD card is inserted after printer turned on without SD card, have either to do a "refresh" or "attach media" from LCD screen menu to see contents of the SD card. 

Also, noticed another bug about Z homing.  Z does not home if using Z probe for Z homing. When trying to home Z (either manual home Z alone, or auto home Z as an initial operation prior to bed leveling) the operation was not carried out by the printer. The LCD shows "Z probe past bed".  If use Z endstop then Z can home normally.

sjasonsmith, 2020-09-13:
For the SD Card, try enabling `NO_SD_HOST_DRIVE` in `Configuration_adv.h`.

Hokeric, 2020-09-13:
Tried enabling NO_SD_HOST_DRIVE in Configuration_adv.h, the there was compilation error, as follows.
FYI, SD card insertion/removal could be auto detected in release version 2.0.6 with NO_SD_HOST_DRIVE disabled.

Marlin\src\HAL\STM32F1\onboard_sd.cpp:29:23: error: 'ON_BOARD_SPI_DEVICE' was not declared in this scope
   SPIClass OnBoardSPI(ON_BOARD_SPI_DEVICE);
                       ^~~~~~~~~~~~~~~~~~Compiling .pio\build\mks_robin_mini\src\src\HAL\shared\eeprom_if_i2c.cpp.o

sjasonsmith, 2020-09-13:
Add this somewhere. If it fixes it, it needs to be added to the pins file. For now you can put it there, or in one of your configuration files.

`#define ON_BOARD_SPI_DEVICE  1`

Hokeric, 2020-09-14:
Adding #define ON_BOARD_SPI_DEVICE 1 to the pins file fixes the SD card issue. Now the SD card insertion/removal can be detected automatically like it is in the earlier 2.0.6 release.

BTW, the Z homing issue mentioned above is no longer observed. The "Z homing bug" mentioned might be due to my handling error. Please ignore that.

Hokeric, 2020-09-15:
Probably should leave it open until changes are updated in the official config files.

Hokeric, 2020-09-25:
I am closing this case as the change has been updated in the latest bugfix release.

closed 2020-09-25 by Hokeric