MarlinFirmware/Marlin #19361: [BUG] [2.0.6.6Cannot Mount Probe

FlailAway, 2020-09-12:
### Bug Description
Corexy home-built. Marlin 20006
No room within printer frame to mount Probe

### Steps to Reproduce
When at Y = 0, there is a front rail of the printer frame about 10mm in the -Y direction
I have the Probe offset at (0, -30, -17.6);
I have "Pause to mount" set
At the Pause, it raises Z higher than the Probe clearance needed. All good.
But, it does not move the Y back so I cannot fit the Probe.

**Expected behavior:** [What you expect to happen]
When it raises Z for the Probe mounting, also move back the Y and X (if any) by the offset amounts PLUS the Probing Margins, if any set.

boelle, 2020-09-12:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

FlailAway, 2020-09-13:
OK, downloaded the latest bugfix you linked and G28 homes X, Y, Z and then G29 raises the Z to 27.6 but stays put at X = 0, Y = 0 and Pauses with the "Mount probe," (or whatever) on the screen.

I have not touched "c..._adv" to save you some time.
[corexy-probe.zip](https://github.com/MarlinFirmware/Marlin/files/5214675/corexy-probe.zip)

FlailAway, 2020-09-13:
Actually,just thinking about it some more, what about it moving to the first probing point but at the elevated X? The steppers should hold it well enough for magnet mounted probe. Even if there is slight X-Y movement putting the probe in place,it really should not matter to the end result.

ellensp, 2020-09-14:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

FlailAway, 2020-09-14:
As well may be, BUT this is a simple issue to test for yourself. That is why I did not bother including the zipped files.

You can test it with very little effort. Set that offset to something, anything for the Y and Z then do a G28+G29 and it will stay pout at the X and Y and raise the Z.

OR I could waste half an hour of my time doing your preferred process for you. I am just reporting what is happening. If that is not good enough for SIMPLE issues then no point mentioning them.

Your approach is a little overly autocratic with my previous thread, you closed it without offering any alternative. Users too would appreciate a complete procedure to be followed.

FlailAway, 2020-09-15:
Oooops, MY mistake but you guys also missed it and the ensuing Storm in a Teacup prevailed.

In the OP I showed, "**I have the Probe offset at (0, -30, -17.6);**"

Note the MINUS-30, caint be done with **Soft Endstops** active.

Lesson for us all can be learned here, I guess. :)

closed 2020-09-15 by FlailAway

sjasonsmith, 2020-09-15:
Perhaps the procedure @ellensp recommended wouldn’t be useful here, but multiple issues similar to this show up every day, and you can’t expect a developer to dedicate hours of their time to each one unless there are real bugs to work out. At a glance it seemed like a bed leveling/probing issue, so it got that response.

In your case I don’t even see a bug. You are just expecting Marlin to magically know the physical constraints of your printer without configuring it accordingly. There is probably a configuration option or variety of configuration options which could help solve your problem.

Given that, this is the generic response you probably should have received:

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).