MarlinFirmware/Marlin #19362: [BUG] problem with build firmware SKR 1.3

joneggg, 2020-09-12:
Hello, when trying to flash skr 1.3, it gives this:

[image]

Windows 10, Latest VSCode, latest platformio, github desktop ,marlin the old one that I used with the first firmware (august 2019)

can this be repaired somehow?

ellensp, 2020-09-12:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

boelle, 2020-09-12:
you are missin a package for vscode/platform.io (red text)

joneggg, 2020-09-12:
I just download latest marlin and problem solved

closed 2020-09-12 by joneggg