MarlinFirmware/Marlin #19363: [BUG] MKS Robin E3

tassiolucas, 2020-09-12:
### Bug Description

I own an MKS Robin E3 and have tested all possible firmware. Both the release tag and the bugfix make the flash but crashes on upload 100%. I used the minimum settings of MKS Robin E3 and only got support in third party forks. Sad

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5213018/Configuration.zip)

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Make any configuration using mks_robin_e3 in the environment.
2. Place the SERIAL 1 port for this environment
3. Upload the code using the standard Platformio procedure
4. When flashing the firmware on MKS Robin E3 V1.0 it uploads strangely faster and crashes at 100%

**Expected behavior:** Function normally as it is in the old version of third party repository - https://github.com/erezagami/Marlin/tree/2.0.x-bf-makerbase-256k

**Actual behavior:** Hangs at 100% upload

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2020-09-12:
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

thisiskeithb, 2020-09-12:
Your config required some changes to get working with the latest `bugfix-2.0.x` (1917ed8), but I tried it on my MKS Robin E3/stock Ender-3 controller and I cannot replicate the issue.

Try formatting your SD card and make sure it's FAT32. Also, try using another SD card.

sjasonsmith, 2020-09-13:
Mine seems to update fine as well, using `bugfix-2.0.x` (7c7e663911f859b511eaf5066e7c331f59148704).

I don't have a screen connected. Do these boards actually show a progress indicator when updating firmware?
I followed the instructions on the [MKS Wiki ](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/Marlin_firmware)page, which instructs you to place Robin_e3.bin on an SD card.

@tassiolucas are you doing something different?

thisiskeithb, 2020-09-13:
> Do these boards actually show a progress indicator when updating firmware?

They do.

tassiolucas, 2020-09-13:
Guys,
I've tried all versions of 2.0.1 onwards, which are the ones that support the MKS Robin V1.0 card all day. The only versions that work are from third party repositories. Like https://github.com/makerbase-mks/MKS-Robin-E3-E3D (which version 2.0.6 only works if you copy the Marlin buildroot from that indicated version, and also from other third parties https: // github.com/erezagami/Marlin/tree/2.0.x-bf-makerbase-256k that work only with some bizarre screen bugs.

All attempts were made with default platformio.ini settings

default_envs = mks_robin_e3
#define MOTHERBOARD BOARD_MKS_ROBIN_E3
SERIAL_PORT 1

[image]

sjasonsmith, 2020-09-13:
Did you try the `bugfix-2.0.x.zip`?

tassiolucas, 2020-09-13:
Yes... Yes, again, I already tried the bugfix version, from 2.0.6.1 to 2.0.1. No version of this boot repository after 100% upload.
I've tried to use even the recommended Configuration.h, nothing works. Remembering my sign:
MKS Robin E3 V1.0
[image]
[image]

thisiskeithb, 2020-09-13:
There’s likely some kind of hardware issue because we can’t replicate the problem on our end.

ellensp, 2020-09-13:
@thisiskeithb let them try your working bin?  perhaps something wrong with their build environment.?

thisiskeithb, 2020-09-13:
Try this bin based on the latest `bugfix-2.0.x` (d93471f) using the stock Ender-3/MKS Robin E3 config from the [Marlin Configs repo](https://github.com/MarlinFirmware/Configurations/tree/45036a755ed3dfe4ec15c70851a52de447ef159b/config/examples/Creality/Ender-3/MKS%20Robin%20E3%201.0):

[Robin_e3.zip](https://github.com/MarlinFirmware/Marlin/files/5213691/Robin_e3.zip).

Here's a bin based on the same `bugfix-2.0.x` tweaked to allow for tassiolucas's config to compile since it's out of date: [Robin_e3.tassiolucas.zip](https://github.com/MarlinFirmware/Marlin/files/5213694/Robin_e3.tassiolucas.zip)

Both bins flash fine on my MKS Robin E3 V1.0 with `CR10_STOCKDISPLAY`.

Again: Try formatting your SD card and make sure it's FAT32. Also, try using another SD card. If that doesn't work, it's likely a hardware issue.

tassiolucas, 2020-09-13:
Ok. The bugfix-2.0.x version only worked with this configuration of Ender-3 / MKS Robin E3 1.0 / - As well as the 2 Bins files sent by your compilation [Robin_e3.zip. and Robin_e3.tassiolucas.zip].

I greatly appreciate your support @thisiskeithb as well as everyone's help.

Unfortunately with the infinity of documentation and the manufacturer's poor clarity on which configuration to follow it was unable to flash, and the upload time was strangely fast compared to the .bin that worked.

My printer is the Ender 3 PRO and as this configuration was in the Ender 3 folder, I couldn't find it, only SKR's E3 V1 (which is not my card)

I don't know if you can help me too. But I'm trying to install 3D Touch on the PB1 port on the card. And I can't seem to get the sensor or calibration to work. It always goes to the limit of the Z axis until it triggers the endstop. Failing to measure the distance with the table.

I'm using this [tutorial](https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/3D_TOUCH)... however I've seen several videos and nothing helps me. I don't know if it's the door that doesn't work or the pin positions that aren't right.

thisiskeithb, 2020-09-13:
Glad those worked, but this Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2020-09-13 by thisiskeithb

tassiolucas, 2020-09-13:
OK! Thank you very much for your help and agility!