MarlinFirmware/Marlin #19366: [FR] Add white frame to dwin display selection

jdogmcsteezy, 2020-09-12:
### Description

Add white frame to icon selection for the ender 3 v2 dwin display.  This would make the UI more pleasant and easier to use.  Currently, the selection is difficult to see if not looking from directly above it.  I provided a video below that clearly shows the problem.

### Feature Workflow

1. When printer is turned on the menus Print Icon will have a white box around it indicating it is selected.
2. The rotary encoder is turned to select Prepare Icon and a white box is displayed around it indicating it is selected.
3. Print Icon white box is erased indicating it has been deselected. And so on . . .

#### Additional Information

* The icons that need this behavior are the: Print, Prepare, Control, Info, Level, Pause, Continue, Stop, Tune.
* Someone has compiled Marlin with this feature but has not opened sourced it: https://smith3d.com/ender-3-v2-bltouch-firmware-installation-guide-by-smith3d-com/
* I have written the code and will be issuing my first PR.
* Here is a video of the problem: https://streamable.com/5uzmsv
* Here is a video of the solution: https://streamable.com/af00v1

jdogmcsteezy, 2020-09-12:
#19368

CRCinAU, 2020-09-13:
Nice first PR - thanks for the contribution - saves being 'yet another thing' for me to hunt down :)

thisiskeithb, 2020-09-13:
PR has been merged. Closing.

closed 2020-09-13 by thisiskeithb

jdogmcsteezy, 2020-09-13:
@CRCinAU  Thanks!  Really happy to be able to contribute!  If I have any questions about the DWIN code for further improvements who should I ping?

CRCinAU, 2020-09-14:
@jdogmcsteezy - Best place is probably the discord channel: https://discord.gg/uZAfgf