MarlinFirmware/Marlin #19370: Recent LPC176x SPI causes cause pin conflicts on MKS SGEN_L with FYSETC Mini display

sjasonsmith, 2020-09-13:
### Bug Description

Recent changes to LPC176x SPI code in commit 1a4b82a5d663ab7f68199bf77183047c1133e479 configures pins for all SPI controllers, even if they are not used.

This causes problems for boards which do not use all SPI controllers, and repurpose those pins for something else. If they are only used as GPIO, they will likely not reconfigure their pin function, and the pin will remain connected to the SPI controller.

I discovered this on an MKS SGEN_L when using a FYSETC Mini display. It looks like there is potential for additional conflicts when other displays are used.

The problem occurs due to this initialization code in `SPIClass::SpiClass`, because the same pin (`P0_15`) is both `BOARD_SPI1_SCK_PIN`, and the `LCD_RESET_PIN` pin for the FYSETC Mini display.

```
  PINSEL_CFG_Type PinCfg;  // data structure to hold init values
  #if BOARD_NR_SPI >= 1
    _settings[0].spi_d = LPC_SSP0;
    _settings[0].dataMode = SPI_MODE0;
    _settings[0].dataSize = DATA_SIZE_8BIT;
    _settings[0].clock = SPI_CLOCK_MAX;
    // _settings[0].clockDivider = determine_baud_rate(_settings[0].spi_d, _settings[0].clock);
    PinCfg.Funcnum = 2;
    PinCfg.OpenDrain = 0;
    PinCfg.Pinmode = 0;
    PinCfg.Pinnum = LPC176x::pin_bit(BOARD_SPI1_SCK_PIN);
    PinCfg.Portnum = LPC176x::pin_port(BOARD_SPI1_SCK_PIN);
    PINSEL_ConfigPin(&PinCfg);
    SET_OUTPUT(BOARD_SPI1_SCK_PIN);
```
### My Configurations

This is a minimal configuration for an MKS SGEN_L 1.0 with a FYSETC Mini 1.2 display.
If it has not already been merged, you will need to include the change from #19369 to fix an issue in the MKS_SGen_L pins file.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5213445/Configuration.zip)

### Steps to Reproduce

1. Power on board

**Expected behavior:**
    Marlin splash screen, followed by normal menus.

**Actual behavior:**
    Screen visibly powers on, but never displays anything.

sjasonsmith, 2020-09-13:
I have already talked to @rhapsodyv and he is planning to address this.

sjasonsmith, 2020-09-13:
@nvumnov, this is one of the problems preventing your display from working. There is no need for you to report a separate issue now.

rhapsodyv, 2020-09-13:
Working on that! Send a PR soon.

nvumnov, 2020-09-13:
Hello!
Thank you so much for your help. I did not expect that this topic would become relevant so quickly.
I spent a lot of time finding a solution. After replacing the electronics, the printer is worthless. I look forward to solving the problem of the mks SGEN L + Fysetc Mini 12864 V2.1 bundle.
Respect, Nikolai!

sjasonsmith, 2020-09-13:
@nvumnov unfortunately there seem to be more issues with the configuration for this board. When we fix things for one display it breaks for others. It's going to take some more time to figure things out.

rhapsodyv, 2020-09-15:
I think this could be closed, right?

sjasonsmith, 2020-09-15:
@nvumnov can you try this out to see if it is working? You will need the absolute latest bugfix-2.0.x code.

nvumnov, 2020-09-15:
> @nvumnov можешь попробовать это, чтобы увидеть, работает ли оно? Вам понадобится абсолютно последний код bugfix-2.0.x.

Hello!
Last night I tried bugfix-2.0.x. No image. After the backlight color test, nothing appears on the screen when loading, after which the backlight changes to red and everything.
Any suggestions to change?

sjasonsmith, 2020-09-15:
I need to test this again. There were several issues related to this, and I don’t think I tested the final product with the FYSETC display. I won’t be able to do this for another day.

@nvumnov can you re-initialize your EEPROM? A bad contrast setting may hide the display. Over a serial port send the commands `M502` followed by `M500`.

nvumnov, 2020-09-16:
Hello!
Yesterday I tried to download the last bugfix-2.0.x again, compiled and everything turned out. The fixes in the HAL_SPI.cpp file were suitable for my electronics kit.
Thank you so much for your help!!!

PR #19372 (2020-09-14): Only setup SPI pins if needed
### Description

It fixes #19370. Now, the SPI pins are only initialised once and just if used. The pin setup is done in the first use.

### Benefits

Fixes #19370.
closed 2020-09-16 by sjasonsmith