MarlinFirmware/Marlin #19374: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?

thorstensd, 2020-09-13:
Hello,

I´m having a strange issue with Marlin-Bugfix 2.0.x.
I like to use E0+E1 with different stepper drivers (SKR 1.4 Zurbo Board)

Thats my config:

#define DISTINCT_E_FACTORS
#define DEFAULT_AXIS_STEPS_PER_UNIT    { 80, 80, 400, 100, 500 }
#define DEFAULT_MAX_FEEDRATE          { 500, 500, 20, 25, 25 }
#define DEFAULT_MAX_ACCELERATION      { 3000, 3000, 100, 10000, 10000 }

#define X_DRIVER_TYPE  TMC2208
#define Y_DRIVER_TYPE  TMC2208
#define Z_DRIVER_TYPE  TMC2208
#define E0_DRIVER_TYPE TMC2208
#define E1_DRIVER_TYPE A4988

#define EXTRUDERS 1

If I compile, I get this error:
In file included from Marlin\src\HAL\LPC1768\../../core/../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\../../core/serial.h:24,
                 from Marlin\src\HAL\LPC1768\DebugMonitor.cpp:25:
Marlin\src\HAL\LPC1768\../../core/../inc/SanityCheck.h:2487:35: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2487 | static_assert(COUNT(sanity_arr_1) <= XYZE_N, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:2487:35: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2487 | static_assert(COUNT(sanity_arr_1) <= XYZE_N, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\DebugMonitor.cpp.o] Error 1
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\eeprom_flash.cpp:39:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:2487:35: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2487 | static_assert(COUNT(sanity_arr_1) <= XYZE_N, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\HAL_SPI.cpp:51:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:2487:35: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2487 | static_assert(COUNT(sanity_arr_1) <= XYZE_N, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\eeprom_sdcard.cpp:25:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:2487:35: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2487 | static_assert(COUNT(sanity_arr_1) <= XYZE_N, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\eeprom_wired.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:2487:35: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2487 | static_assert(COUNT(sanity_arr_1) <= XYZE_N, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\HAL.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_flash.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_wired.cpp.o] Error 1
In file included from Marlin\src\HAL\LPC1768\../../sd/../inc/MarlinConfig.h:40,
                 from Marlin\src\HAL\LPC1768\../../sd/cardreader.h:24,
                 from Marlin\src\HAL\LPC1768\main.cpp:38:
Marlin\src\HAL\LPC1768\../../sd/../inc/SanityCheck.h:2487:35: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)
 2487 | static_assert(COUNT(sanity_arr_1) <= XYZE_N, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements. (Did you forget to enable DISTINCT_E_FACTORS?)");
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\main.cpp.o] Error 1

ellensp, 2020-09-13:
Please attach actual Configuration files, so we can drop into our test environments without any guess work.

thorstensd, 2020-09-13:
Hello,

please find the config files attached.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5214018/Marlin.zip)

best regards

ellensp, 2020-09-13:
you cant have #define EXTRUDERS 1 and DISTINCT_E_FACTORS

ellensp, 2020-09-13:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-09-13 by ellensp

thorstensd, 2020-09-13:
Ok, thank you very much for your feedback - I had that already, but if I use Extruders 2, I will fail on:
TEMP_SENSOR_1 is required with 2 or more HOTENDS

But I dont have 2 hotends - I´m will use a Y splitter.

Vertabreak, 2020-09-13:
uncomment  #define SINGLENOZZLE

thorstensd, 2020-09-13:
I have found it, #define SINGLENOZZLE need to be enable - my fault.
Thank you very much.