MarlinFirmware/Marlin #19377: Ender 3 v2 LCD screen western characters.

kyraxyg, 2020-09-13:
I have compiled the lastest bugfix for Ender 3 v2 with LCD language turkish. Although I know Ender 3 v2 screen support is work in progress, I want to inform that turkish characters like "ç", "i", "ğ" are shown in Chinese if the source is compiled with LCD screen = tr.

boelle, 2020-09-13:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

kyraxyg, 2020-09-13:
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/5214684/Config.zip)
Attached the Configuration.h and Configuration_adv.h files as zip

sjasonsmith, 2020-09-13:
I think that display uses GB2312 encoded text. Many strings won't display properly once you are outside of normal 7-bit ASCII characters.

One of the users on here has commented about this in the past, but I can't seem to locate their name or comments at the moment. I don't know whether there is a way to adjust the display's behavior, or if the only solution is to transcode the strings to GB2312.

enigmaquip, 2020-09-14:
That was me @sjasonsmith 
I haven't had a chance to do further testing (ie sending gb2312 encoded strings to see if I get what I expect), that info is strictly from the documentation for the dwin chip in the creality screen. From what I can tell there's no way to change that and we'll just have to encode to gb2312

closed 2020-10-22 by github-actions[bot]