MarlinFirmware/Marlin #19385: [BUG] UBL not moving the probe in XY-plane while probing the mesh

sverd, 2020-09-13:
### Bug Description
After quite some conversations with sjasonsmith on Discord, he also could not find any problem with the configuration used and he advised me to report here.  It is about the following problem:
When trying to have UBL probe some mesh points automatically, the nozzle/probe is "homed" and all probing takes place at the "home location" (no movement in XY but expected behavior in Z). As in my case the home position is next to the bed, only probe errors are obtained (which makes sense). The log indicates a single movement to the first point, but this did not happen with the actual printer.

### My Configurations
My original configuration is quite extended so for convenience a simpler configuration was created that also has the same problem, but is as close to the default configuration as I could get. Test was done with bugfix-2.0.x as of today.
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5215080/Configuration_adv.zip)

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Start device
2. load and save defaults
3. find the home position
4. Start probing at mesh point (G29 P1)

**Expected behavior:** [What you expect to happen]
The printer homes again, moves to the first point to be probed and starts probing. Followed by the second point etc. until the mesh is finished.

**Actual behavior:** [What actually happens]
The printer homes again, and starts probing at the "home-position" for all mesh points, so it does not move in the X and/or Y direction. The probing movement in the Z-direction is as to be expected, but when the home-position is next to the bed, only probing failures will occur (which is logical).
After all points are finished the tries to move the bed in the negative direction which is impossible because it was still in the home position.

#### Additional Information
The debug messages from Pronterface give more information what is happening. It does indicate that a single step was taken, this was however not seen on the actual printer
[log-basic-UBL-16microsteps.txt](https://github.com/MarlinFirmware/Marlin/files/5215090/log-basic-UBL-16microsteps.txt)

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2020-09-14:
@Vertabreak indicated on Discord that he things the configuration is invalid for UBL, because the margins/insets don’t leave much for UBL to probe with only a 3x3 grid.

That is a valid concern, although that still leaves me troubled be the behavior for three reasons:
1. It should NEVER probe outside of the probably area, and this is probing off the bed. I couldn’t find a reason in the configs that would cause this.
2. When the first probe failed, it then claimed to move to point 2 (with a different XY) without actually moving.
3. It continues probing all 9 times, even though each probe failed. It seems there is something really not right in the error path here.

closed 2020-10-22 by github-actions[bot]

RandyL00123, 2020-12-01:
Did you ever find an answer? I'm running into the same problem.

sverd, 2020-12-03:
Yes and no. This happened after a complete upgrade of my printer (8 to 32 bit controller, different hotend etc.). In the end I got it working correctly but a few things might have been the cause of the problem and I could not pinpoint it (I needed the printer the work so when it finally worked I started printing immediately).
So what did I do (many times):

- I started from the 8-bit configuration assuming the settings related to the hardware would simply transfer to the new. However it seems at a certain point some velocities were so high that the hardware could not keep up and complete steps were skipped (or I felt the need to use the emergency reset). A restart complete from scratch with the values seemed to fix this 

- it seems the meaning and/or names of configuration variables seems to slightly change sometimes. First I was transferring as many settings as possible to try to speed things up. In the end I started also with the confgurations from scratch with the bare minimum of options enabled. Even functions that were working perfectly on the 8-bit system were disabled first and enabled one-by-one. Magically this led slowly to a system with the options I wanted enabled.
- For the inset margins of the UBL mesh start with a 3x3 start in the middle of the bed. Then change the insets/margins one at a time to see their effect and to optimize. This will give some idea of their effect and where to stop.

All-in-all I cannot give you the final answer, but restart from scratch and try each setting one at a time. If you keep an existing configuration file at hand, copying those one by one may speed things up slightly.  Do not copy blindly however!
Hopefully this helps a little bit

sverd, 2021-01-27:
After some small modification I decided to compile the latest Marlin again, and again I ran into this problem. After quite some time puzzling I found a "working solution" with the following two adjustments:

- I reduced the number of microsteps on both the X and the Y driver with a factor of 2 (took half the nr of microsteps). I think this reduces the nr of needed ISR's quite significantly (factor of 4??) so the microcontroller can keep up speed more easily.
- I put a small piece of filament in the filament sensor while performing the G29 (I have the habit of keeping my filament stored while setting up the system. It seems however that an empty filament sensor may result in a "probe triggered early" error. So putting a piece of filament in the sensor works around it. (disabling the filament sensor may also work, I did not test that)