MarlinFirmware/Marlin #19388: [BUG] G28 not recognizing endstops, homing via LCD menu works

runruh, 2020-09-14:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

This is specifically for the new Creality 4.2.7 mainboard and an Ender 3. I took the default configuration files for the Ender 3 Pro v1.5 along with the status and boot screens from ender v1.

Changes made from default configuration:

- set mainboard to BOARD_CREALITY_V427
- turned on custom boot screen and status screens
- updated author line
- esteps for extruder from 93 => 415
- enabled s curve acceleration
- disabled arc support
- turned on lcd option for bed leveling (including center)
- changed serial baud rate to 25000

** Steps to reproduce:
1. Issue a G28 command via USB tethered machine (octoprint in my case)

**Expected behavior:** G28 command homes successfully, same as using LCD

**Actual behavior:** G28 triggers motion, but endstop triggers do not stop motion, have to turn off machine

#### Additional Information

* Relevant configuration files: [Archive.zip](https://github.com/MarlinFirmware/Marlin/files/5215509/Archive.zip)

ellensp, 2020-09-14:
12 days ago a bug was found that if there was no SD-card inserted in the SD-card slot the endstops could stop working

Put a SD card into controllers sd card slot see if the issue persists.

This is fixed is only in bugfix so far.

runruh, 2020-09-14:
> 12 days ago a bug was found that if there was no SD-card inserted in the SD-card slot the endstops could stop working
> 
> Put a SD card into controllers sd card slot see if the issue persists.
> 
> This is fixed is only in bugfix so far.

Confirmed this does fix my issue. Reference issue: [#19223](https://github.com/MarlinFirmware/Marlin/issues/19223)

runruh, 2020-09-14:
Referenced fix is in this pull request: #19226

closed 2020-09-14 by runruh