MarlinFirmware/Marlin #19390: Bltouch v3.1 probing but not leveling

ase95, 2020-09-14:
### Bug Description

my bltouch v3.1 is probing but my first layer not even near being level, i have pre leveled the bed manually.
i've checked my axis steps but even after adjusting them,it's not working... (im using marlin 1.8 since im having issue with the 2.0 and infinite loop of the bootscreen)

My printer is a Tevo Tornado

My starting code: 
G28 ; home all axes
G29; Auto Bed Levelling
G1 Z0.2 F3000 ; get ready to prime
G92 E0 ; reset extrusion distance
G1 X160 E15 F600 ; prime nozzle
G1 X180 F5000 ; quick wipe

[Here some pictures to see what im talking about ](https://github.com/MarlinFirmware/Marlin/files/5218641/pictures.zip)

### My Configurations

[config.zip](https://github.com/MarlinFirmware/Marlin/files/5218604/config.zip)

### Steps to Reproduce
im just printing a bed level test to see if it's leveled or not.

**Expected behavior:** [What you expect to happen]
expecting to have an even layer on all the printing bed

**Actual behavior:** [What actually happens]
looks like bltouch its not even on

boelle, 2020-09-14:
marlin 1.8 is no longer active worked on, 
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

ase95, 2020-09-14:
> marlin 1.8 is no longer active worked on,
> Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

i would use the 2.0 if i could, but for some reason my printer gets stuck on the bootscreen,other people tried my config and it works fine on them

Azrael321, 2020-09-14:
I think there is missing 

`M420 S1`

in the start code

ase95, 2020-09-14:
> I think there is missing
> 
> `M420 S1`
> 
> in the start code

should put it right after g29?

mustang5269, 2020-09-14:
He wouldn't need that if he has restore leveling after g28 enabled in his firmware correct?

Azrael321, 2020-09-15:
Yes, should be somewhere after G29. If you use G28 after this command, command is normaly deactivated again.

Yes, if you have restore leveling after g28 enabled you should not need this command

oliver-eifler, 2020-09-15:
if you use G29 after G28 in your start g-code neither M420 nor restore leveling (it only restores leveling state not data) in firmware is needed...G29 probes the mesh and activates auto bed leveling.

There was extra code for bl-touch 3.x added in marlin 1.1.9 ... AFAIK bl-touch 3.1 wouldn't work with marlin 1.1.8
You should use at least latest marlin 1.1.9

ase95, 2020-09-15:
> if you use G29 after G28 in your start g-code neither M420 nor restore leveling (it only restores leveling state not data) in firmware is needed...G29 probes the mesh and activates auto bed [...]

THANKS!  i'll start doing the config on the 1.1.9 asap and i'll hope by bltouch finally start working

ase95, 2020-09-15:
still having same issue 😢

Hokeric, 2020-09-17:
> still having same issue 😢

This might sound too obvious, but have you checked Z offset?

ase95, 2020-09-17:
> > still having same issue 😢
> 
> This might sound too obvious, but have you checked Z offset?

sadly i have...im out of ideas,tried so many different firmwares/versions but this issue isn't going away.
Any chance the issue is the bltouch itself? like getting wrong readings (?)

Hokeric, 2020-09-17:
> > > still having same issue 😢
> > 
> > 
> > This might sound too obvious, but have you checked Z offset?
> 
> sadly i have...im out of ideas,tried so many different firmwares/versions but this issue isn't going away.
> Any chance the issue is the bltouch itself? like getting wrong readings (?)

I have not tested it myself, this could be related. If the bed level difference is greater than 3mm then the probe might not have gone low enough during leveling. Try change the -3 to -5 and see how it goes.

#define Z_PROBE_LOW_POINT          -3 // Farthest distance below the trigger-point to go before stopping

fiechr, 2020-09-17:
> if you use G29 after G28 in your start g-code neither M420 nor restore leveling (it only restores leveling state not data) in firmware is needed...G29 probes the mesh and activates auto bed leveling.

I'm sorry, but could you explain what the difference between restoring level state and (level?) data actually is? When I was using Manual Mesh Bed Leveling, I saved the Z-offsets from the G29 procedure with `M500` to the EEPROM and later (before printing) restored them/it with `M501` and `M420 S1`.
Now, using BLTouch, I'm unsure if this works the same way for Auto Bed Leveling? So after running `G29` and a `M500`, can I completely restore the state after the `G29` with `M501` and `M420 S1`? Since I don't want to run a leveling procedure every time I print something. I would prefer running a more thorough leveling procedure once after filament change etc..

ase95, 2020-09-18:
> > > > still having same issue 😢
> > > 
> > > 
> > > This might sound too obvious, but have you checked Z offset?
> > 
> > 
> > sadly i have...im out of ideas,tried so many different firmwares/versions but this issue isn't going away.
> > Any chance the issue is the bltouch itself? like getting wrong readings (?)
> 
> I have not tested it myself, this could be related. I am making a wild guess here. If the bed level difference is greater than 3mm then the probe might not have gone low enough during leveling.
> 
> #define Z_PROBE_LOW_POINT -3 // Farthest distance below the trigger-point to go before stopping
> 
> From your pictures, the front is ok (probed first), the middle looks less ok and the backs look worst. There seems to be a trend from front to back. Try change the -3 to -4 and see how it goes.

difference from front to back should be 1mm max looking at what Z the probe activates.
My #define Z_PROBE_LOW_POINT is set to -2 so i'll try change it to -3 if that doesn't work i'll go with -4

sjasonsmith, 2020-09-20:
I am closing this issue, since this is for Marlin 1.1.x which is no longer developed. I recommend you seek help from the community to create a working Marlin 2.x configuration.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-09-20 by sjasonsmith