MarlinFirmware/Marlin #19391: [FR] S42B and Servo42 closed loop stepper driver timing profile (copy of TB6600)

Quas7, 2020-09-14:
### Description

Low-budget closed loop hybrid stepper drivers are nice toys to play around with and might also add some benefits in some situations. For instance, in my case the main benefit is much cooler running nema17 pancakes for direct extruders while still haveing high current feeding in case of high loads without missing steps (although high load/speed affects the extrusion timing).

Some chinese copycat versions are 
- MKS Servo42B (https://github.com/makerbase-mks/MKS-SERVO42B) and 
- bigtreetech S42B (https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0) 
and all seem to be based at least partly on the Misfittech Smartstepper (https://github.com/Misfittech/nano_stepper)

As a microcontroller is doing the motor control in these budget versions, the communication timing should be relaxed as documented by the suppliers. See for instance FAQ Question 5 (p20): https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/blob/master/42%20Stepper%20Motor%20Closed%20Loop%20Driver%20Control%20Board%20Manual.pdf
Otherwise, you risk missing some steps.

Unfortunatelly, the solution provided in the documentation above affects the timing of all connected stepper drivers and could reduce the performance of high resolution steppers (e.g. high-µsteps on TMC220x, LV8729,...).

A better solution is to use the TB6600 driver timings for the respective hybrid axis, as the TB6600 timing is closely matching the targeted values. See also discussion on S42B issue https://github.com/bigtreetech/BIGTREETECH-S42B-V1.0/issues/14.

Therefore, copying the TB6600 settings to a "hybrid steppers profile" might be worthwhile for an easier setup.

### Feature Workflow

#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define E0_DRIVER_TYPE *TB6600*    // add here *HYBRID* as selectable profile instead

sjasonsmith, 2020-09-20:
The title of this request does not seem to represent your request at all.

Quas7, 2020-09-21:
adapted slightly the wordking but @sjasonsmith please make a better proposal than just stating "does not seem to represent at all"

sjasonsmith, 2020-09-21:
The title is all about the timing profile, but the real request seems to be that you want hybrid timing that doesn’t force all drivers to use the same timing?

Quas7, 2020-09-22:
Oh, ok. That is a misunderstanding.
The independent timing profiles are already there as you can chose stepper drivers for every axis seperately in Configuration.h today. Also the correct timing profile is available as the TB6600 driver timing is fitting perfectly.

We would just need to add an alias that links "HYBRID" to the TB6600 profile in the background.

sjasonsmith, 2020-09-22:
> The independent timing profiles are already there as you can chose stepper drivers for every axis seperately in Configuration.h today.

Yeah, that's not how those work. All steppers will use the timing of your slowest driver.

Quas7, 2020-09-22:
wow, ok - that suprises me a bit. 
I ran some tests up to 300mm/s and only my extruder has a hybrid driver. Running on TB6600 it the movment "sounded" very differently configuring it this way than with setting it globally in Configuration_adv.h.

sjasonsmith, 2020-09-22:
If you specify timings in Configuration_adv.h they will override any timings defined for your declared stepper drivers, even if it violates your driver's specifications.

If you do not specify timings, it selects timings automatically in Conditionals_adv.h. You can see from this code that it will end up selecting the slowest timing of your selected drivers.
https://github.com/MarlinFirmware/Marlin/blob/47aa61c0ec4a4fe511bd7c629ce18d5ea8007da8/Marlin/src/inc/Conditionals_adv.h#L338

sjasonsmith, 2020-09-22:
This is done because all stepping happens in one function. The pulse for all steppers starts at the same time, and a single delay is used for that pulse duration. Making this use flexible timing adds complexity to that routine, which would hurt performance for some users.

Quas7, 2020-09-22:
This makes absolutely sense - thanks for providing the details!
I just might have messed up my little audio experiment by using the hybrid spreadCycle stuff in parallel.

I will now optimize the global settings accordingly just for fun although it does really not look like as there is much real performance impact with those us and ns timings. ;)

closed 2020-09-22 by Quas7