MarlinFirmware/Marlin #19392: [FR] Kingroon KP3 - Suggest to use PIN PB2 (Reserved Connector) instead of PA8 (WiFI IO0) for BLTouch

Hokeric, 2020-09-14:
### Description

Suggest to use PIN PB2 (Reserved Connector) instead of PA8 (WiFI IO0) for BLTouch. This will have the following benefits:
1. Enable the use of WiFi Module and BLTouch at the same time
2. The Reserved Connector is a standard connector which is easier to connect to and connection is more secure
3. User does not need to change the PINS file to use PB2 for every new release

### Feature Workflow

1. [First Action] Nil
2. [Second Action] Nil
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2020-09-16:
There isn't enough information here to know what you need. Kingroon doesn't appear in the Configurations repo or as a board name. What MOTHERBOARD does it use?

To avoid breaking backward compatibility with existing users, you could do something like this, which enables overriding the pin from your configuration file:

```
#ifndef SERVO0_PIN
  #define SERVO0_PIN                          PA8 // (WiFI IO0 pin)
#endif
```

sjasonsmith, 2020-09-16:
I intended to suggest that you make the change above (or something similar) and contribute it back to Marlin through a Pull Request.

Hokeric, 2020-09-16:
Thank you for your response. The board is MKS Robin Mini. 

The change will be as follows:

Current:
#ifndef SERVO0_PIN
  #define SERVO0_PIN                          PA8 // (WiFI IO0 pin)
#endif

After the suggested change:
#ifndef SERVO0_PIN
  #define SERVO0_PIN                          PB2 // (signal pin of the reserved box connector)
#endif

I can do the pull request. Could you advise the procedure? I have not done it before.

sjasonsmith, 2020-09-20:
I would advise to leave the current pin. Changing it is going to break anybody that is already depending on the existing behavior.

Another option would be to disable both (since the board has no labeled servo connection) and require the user to specify one.
This will still break builds, but will not start using another pin which might already be used for something else.

```
#ifndef SERVO0_PIN
  // This board has no dedicated servo pins. Possible options include:
  // #define SERVO0_PIN PA8 // (WiFI IO0 pin)
  // #define SERVO0_PIN PB2 // (signal pin of the reserved box connector)
#endif
```

There are many different workflows that can be used to create a pull request. Here is one which is documented on the Marlin site:
https://marlinfw.org/docs/development/getting_started_pull_requests.html

Hokeric, 2020-09-21:
I like the idea of giving options for the servo pin using PA8 or PB2 without enabling either one by default. Although this might break builds of users already using the PA8 pin, it can be easily fixed by simply re-enabling the PA8 pin in the pins file, or define it in the config files. 

That is, I think the suggestion given above (copied below with addition of info about the J11 connector) is the best solution.

#ifndef SERVO0_PIN
  // This board has no dedicated servo pins. Possible options include:
  // #define SERVO0_PIN PA8 // (WiFI IO0 pin)
  // #define SERVO0_PIN PB2 // (signal pin of the reserved box connector - J11)
#endif

The PB2 pin is a better and easier solution in my opinion as it is not being for anything and the box connector is easier to connect than the WiFi connector. Therefore, it will be good as alternative to PA8.

FYI, I did previously try to re-define SERVO0 pin in my config file to use PB2 without disabling the PA8 in pins file, but somehow BLtouch did not work. I had to specifically disable the "define SERVO0_PIN PA8" for "define SERVO0_PIN PB2" to work.

sjasonsmith, 2020-09-21:
The `#ifndef` will allow it to be overridden. That was not there before.

Hokeric, 2020-09-21:
Ah, I see. Thank you.

Hokeric, 2020-09-25:
After some further thought, may I suggest making change as follows. 

Before change:
  #define SERVO0_PIN PA8 // Enable BLTOUCH support on IO0 (WIFI connector

After change:
#ifndef SERVO0_PIN
  #define SERVO0_PIN PA8 // Enable BLTOUCH support on IO0 (WIFI connector
#endif

This might be the best solution as it will not break builds currently using PA8, and at the same time provide flexibility to users who want to use other pins such as PB2 for BLTOUCH without touching the pins file. Users who prefer using PB2 (or other pins) can re-define SERVO0 in one of their config files as follows which will override the definition in pins file.

   #define SERVO0_PIN PB2

Thank you.

Hokeric, 2020-11-09:
The released 2.0.7 still have this issue which breaks compilation using self-defined SERVO0_PIN.
Could you please change the SERVO0_PIN to conditional def?

Before change:
#define SERVO0_PIN PA8 // Enable BLTOUCH support on IO0 (WIFI connector)

After change:
#ifndef SERVO0_PIN
#define SERVO0_PIN PA8 // Enable BLTOUCH support on IO0 (WIFI connector)
#endif

sjasonsmith, 2020-11-12:
Is this the pin you are using on your board (in yellow)?
[image]

Hokeric, 2020-11-12:
Yes, that is address PB2.

Hokeric, 2020-11-12:
> Is this the pin you are using on your board (in yellow)?
> [image]

Yes, I am using this connector J11. It is pin PB2.

Hokeric, 2020-11-16:
Thank you very much for your help. Much appreciated.

sjasonsmith, 2020-11-17:
This is merged. Given the simplicity of the change I'll go ahead and close this issue.

closed 2020-11-17 by sjasonsmith