MarlinFirmware/Marlin #19396: [FR] Flow value edit option

Liger0, 2020-09-14:
### Description

Having a value to set when not printing, so not in the tuning menu, would be useful.
For instance, the current implementation making it exist only in the Tune option while printing is hard to follow to me.
I was having issues with prints overextruding I could not figure out, and it took me a long time to realize it was because of the Flow being set and not changeable unless you are printing. 
Sometimes I do test prints and set values like current, Z offset and flow on the fly, and Store the settings when still printing.
The fact is I did not realize the flow was also saved, because, seeing it was not editable while not printing made me think it isn't stored along with other options, which would make sense since it's a thing to set in the slicer anyway.
I don't look at the Tune options unless I am still calibrating the printer, as I put final values only when not printing as I am done with all the test prints, which is why Tune exists in the first place, I suppose.
Being forced to do a thing while printing, especially when it's not supposed to be a thing to use permanently and expecting to look at Tune values while everything beforehand looked fine on the menu is not ideal to me.

### Feature Workflow

1. Use the LCD while not printing
2. Set flow percentage

rhapsodyv, 2020-09-14:
Maybe you should try calibrate your extruder. In my experience, if I need to change the flow often, is because other thing is wrong with the printer.

Liger0, 2020-09-14:
I think I may have spoken too early, it was the gcode autogenerator placing the extrusion multiplier automatically this time...

closed 2020-09-14 by Liger0