MarlinFirmware/Marlin #19401: [FR] Assisted Build Area Measurement

qwewer0, 2020-09-15:
### Description
A new `M___ X Y Z` command that would move (home) one or more axis to the homing position and would provide the traveled distance in mm to the serial connection.

e.g.: If prior to the command, we move one or more axis (let's say X axis for this) to the opposite end of the homing position, then by sending `M___ X` to the printer, the command would move (home) the X axis and after it reaches the homing position, it would provide the traveled distance in mm, which then can be used to determine `X_MAX_POS`.

I personally could/would use this feature to adjust my printers `PROBING_MARGIN`, to be able to probe as big area on the bed as possible.

### Feature Workflow

1. Connect to the printer. (e.g. Pronterface, Octoprint, etc.)
2. Move one or more axis (that we want to test) to the opposite end of the homing position
3. Send `M___` followed by `X`, `Y`, `Z` (e.g. `M___ X Y Z` to measure X, Y and Z axis)
4. Wait for it to complete the procedure
5. Get value(s) in mm on the connected software

thisiskeithb, 2020-09-15:
Keep in mind that for many printers, the `*_MIN_POS` & `*_MAX_POS` are not the same as printable area as hotends tend to home off the build plate or move beyond `*_MAX_POS`.

It won't be impossible to define the build area, but there would have to be some more steps/menus to confirm when the nozzle reaches/leaves the actual build area.

qwewer0, 2020-09-15:
@thisiskeithb

> Keep in mind that for many printers, the `*_MIN_POS` & `*_MAX_POS` are not the same as printable area as hotends tend to home off the build plate or move beyond `*_MAX_POS`.

Yeah, I know. I have my `X_BED_SIZE` set to `234`, and my `X_MAX_POS` to `269.5`, in order to able to probe my bed area with `PROBING_MARGIN` of `10`, but for that I had to measure it a few times, but with this would have been easier.

> It won't be impossible to define the build area, but there would have to be some more steps/menus to confirm when the nozzle reaches/leaves the actual build area.

I'm not saying that the command would do any setting change, just give use a raw value in mm, then it is up to the user to what to do with it.

daleckystepan, 2020-09-15:
It can be possible using Z Probe if you have properly calibrated offset of the probe and nozzle.

qwewer0, 2020-09-15:
@daleckystepan 

> It can be possible using Z Probe if you have properly calibrated offset of the probe and nozzle.

Yes, if we have the z-offset correctly, then the given values will be useful, but if not, then it still can be a good approximate value.

qwewer0, 2020-09-15:
I thought about a Z axis having a probe. Would it give wrong (slightly off) values if the probe isn't defined in the firmware, and only used as an endstop?

qwewer0, 2020-09-15:
Side note: If I would have to do the measurements "manually" with the printer, then after the printer is homed, I would move each axis with the software endstops disabled, until I think it cannot move anymore or the motor skipped a step, and the nozzle position would be close to the real value that I need, but it would be way easier and more precise with a gcode command, and let the printer do it itself.

qwewer0, 2020-10-01:
This feature might be better to have it integrated inside G28, with a new C parameter as "Count". G28 X Y Z C would be the same as it is already in the description.

closed 2021-02-25 by qwewer0