MarlinFirmware/Marlin #19402: Error compiling in 2.0.6 due to G35.cpp and ASSISTED_TRAMMING with BLTouch present

Mike8-I, 2020-09-15:
When compiling marlin bugfix (//#define STRING_DISTRIBUTION_DATE "2019-07-10")  with assited tramming enabled this error occures: *** [.pio\build\LPC1769\src\src\gcode\bedlevel\G35.cpp.o] Error 1 
error does not occur when assisted tramming is commented out
board: bigtreetech SKR1.4 turbo

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5226876/Configuration.zip)

thisiskeithb, 2020-09-15:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

Mike8-I, 2020-09-15:
I get this error now
In file included from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:42,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/../HAL/LPC1768/inc/SanityCheck.h:28:4: error: #error "framework-arduino-lpc176x package is out of date, Please update the PlatformIO platforms, frameworks and libraries."

I copied my configuration files plus the platformio.ini into the new version... anything wrong with this? platforIO.ide is current version

thisiskeithb, 2020-09-15:
> anything wrong with this?

Yes. Use the `platformio.ini` file from the latest `bugfix-2.0.x` and set `default_envs = LPC1769`.

You can try dragging and dropping your old configs, but you're better off porting your settings to the latest version.

thisiskeithb, 2020-09-15:
Dual hotends is what is tripping up `G35`/`ASSISTED_TRAMMING`. I don't think the two are compatible.

Mike8-I, 2020-09-15:
compiling problem solved. Porting settings is not much appreciated ;-) takes a lot of time....
Dual hotend compatibility to assisted tramming would be much appreciated.

thisiskeithb, 2020-09-15:
Adding the code in the comment linked above allows for your config to compile, but I have no way to test it.

boelle, 2020-09-15:
> takes a lot of time....

not if you use a compare tool, on windows winmerge is a great tool for this

Mike8-I, 2020-09-15:
does compile, but assisted tramming does nor work. 
(started from menu) It first does a complete homing. Then a point at the front left corner. Then the z homing point again. Then it just stops. 
Thanks for the help anyway, was worth a try.
@boelle: will try this tool, thks.

thisiskeithb, 2020-09-15:
Are you able to connect via serial/host like Pronterface or OctoPrint? There may be errors in the serial output that can help debug further.

Mike8-I, 2020-09-16:
It does the front left point, then the error output happens, then the homing Z (again), as best as I could follow it.

Recv:17:21:12.013: echo:Active Extruder: 0 (2)
Recv:17:21:31.892: Selected home phase 896 too close to endstop trigger phase 24. Pick a different phase for Z
Recv:17:21:35.496: echo:Active Extruder: 0
Recv:17:21:35.496: X:130.00 Y:118.00 Z:12.80 E:0.00 Count X:20885 Y:18880 Z:32768
Recv:17:21:47.799: G35 failed at point 1 (Front-Right) X190.00 Y20.00
Recv:17:21:47.799: G35 aborted.
Recv:17:21:50.079: echo:Active Extruder: 0 (2)
Recv:17:22:02.770: Selected home phase 896 too close to endstop trigger phase 8. Pick a different phase for Z
Recv:17:22:06.374: echo:Active Extruder: 0
Recv:17:22:06.374: X:130.00 Y:118.00 Z:12.80 E:0.00 Count X:20885 Y:18880 Z:32768

The endstop trigger phase might be because I forgot to comment it, after senorless homing did not work and I reverted to endstops.

thisiskeithb, 2020-09-16:
Disabling `TMC_HOME_PHASE` until you can find the correct homing phase/tune it, should allow you to use `ASSSISTED_TRAMMING`.

I've put in a PR that will allow `G35` to work on multi hotend setups: https://github.com/MarlinFirmware/Marlin/pull/19418(thanks @ellensp for the fix!)

Mike8-I, 2020-09-17:
The  TMC_HOME_PHASE wasnt the reason. After commenting out, it did still not work.
I could now make the second point work, by significantly putting it closer to the bed center. X= 62 is the biggest number that works. tramming points 3 and 4 still dont work. I feel that there might be another issue.
 #define TRAMMING_POINT_XY { {  20, 20 }, { 162,  20 }, { 162, 187   }, { 20, 187 } }
// The size of the print bed
#define X_BED_SIZE 195
#define Y_BED_SIZE 205
#define NOZZLE_TO_PROBE_OFFSET { -33, -16, 0 }

G35now:
Recv:21:39:07.796: echo:Active Extruder: 0 (3)
Recv:21:39:07.796: X:130.00 Y:118.00 Z:12.80 E:0.00 Count X:20885 Y:18880 Z:32768
Recv:21:39:25.487: G35 failed at point 2 (Back-Right) X162.00 Y187.00
Recv:21:39:25.487: G35 aborted.
Recv:21:39:43.858: echo:Active Extruder: 0 (3)
Recv:21:39:43.858: X:130.00 Y:118.00 Z:12.80 E:0.00 Count X:20885 Y:18880 Z:32768
Befehle zeigt Befehle, die zum Drucker gesendet wurden. ACK zeigt Bestätigungen vom Drucker, die Üblicherweise un

qwewer0, 2020-09-17:
@Mike8-I So, the problem is now, that it doesn't work with e.g. X=163, or you can't compile the firmware?
If it is the first one, then that is because the nozzle won't leave the "safe" (`_MIN_POS`, `_MAX_POS`) area, so with probe X offset of -33, your max "X" value is 195+(-33)=162.
If it is the second one, then does https://github.com/MarlinFirmware/Marlin/pull/19418 fixed it?

closed 2020-09-18 by thisiskeithb

Mike8-I, 2020-09-18:
It does compile.
I aggree that the probe must stay in the safe area. 
All the higher number points that I defined fullfill this, including a safety markin. (though point one shouldnt work, see further down)
Anyway, point 3 and 4 dont work.
Also the logic seems wrong to me.  As my probe is left of the nozzle and in front of it, the possible max nozzle positions for x and y tramming points should be xmay and ymax. The Probe would then be at xmay+(-33) and ymax+(-16), so well within the bed.
This is according the comment in configuration_advance.h:
<< // Define positions for probing points, use the hotend as reference not the sensor.
  #define TRAMMING_POINT_XY { {  20, 20 }, { 162,  20 }, { 162, 187   }, { 20, 187 } }>>

In my opinion the possible max position should be: xmax - NOZZLE_TO_PROBE_OFFSET, limited to xmax (for negative probe positions). The position for calculating the tramming point should be xTramming point +NOZZLE_TO_PROBE_OFFSET. Same for y.
As I use 20,20 for the first point, and it works and it does this point with the sensor well within the bed(!), something is  not correct here, too. Could it be that the signs for the min and max point calcualtions are the wrong way?