MarlinFirmware/Marlin #19403: [BUG] (Language stuck on Japanese)

Benderrodrigu3z, 2020-09-15:
Marlin 2.0.6.1

I can set the language to English BUT IT DOESNT effect the board language, I have it changed in the code, see attached but its always Japanese....

Benderrodrigu3z, 2020-09-15:
PS I have the Ender 5 Plus with silent board Creality 2.2.1

boelle, 2020-09-15:
did you read the template? (no you did not)

Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Benderrodrigu3z, 2020-09-15:
I did attach them but for some reason there gone now, anyways see attached

Benderrodrigu3z, 2020-09-15:
strange, I can add them but they don't add to the attachment list

Benderrodrigu3z, 2020-09-15:
/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#define CONFIGURATION_H_VERSION 020006

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * https://reprap.org/wiki/Calibration
 * https://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * https://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * https://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * https://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// config/examples/delta directory and customize for your machine.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a SCARA printer start with the configuration files in
// config/examples/SCARA and customize for your machine.
//

[... 2328 lines cut ...]

/**
 * R/C SERVO support
 * Sponsored by TrinityLabs, Reworked by codexmas
 */

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Leave undefined or set to 0 to entirely disable the servo subsystem.
 */
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// (ms) Delay  before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

// Allow servo angle to be edited and saved to EEPROM
//#define EDITABLE_SERVO_ANGLES

Benderrodrigu3z, 2020-09-15:
/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration_adv.h
 *
 * Advanced settings.
 * Only change these if you know exactly what you're doing.
 * Some of these settings can damage your printer if improperly set!
 *
 * Basic settings can be found in Configuration.h
 *
 */
#define CONFIGURATION_ADV_H_VERSION 020006

// @section temperature

//===========================================================================
//============================= Thermal Settings ============================
//===========================================================================

/**
 * Thermocouple sensors are quite sensitive to noise.  Any noise induced in
 * the sensor wires, such as by stepper motor wires run in parallel to them,
 * may result in the thermocouple sensor reporting spurious errors.  This
 * value is the number of errors which can occur in a row before the error
 * is reported.  This allows us to ignore intermittent error conditions while
 * still detecting an actual failure, which should result in a continuous
 * stream of errors from the sensor.
 *
 * Set this value to 0 to fail on the first error to occur.
 */
#define THERMOCOUPLE_MAX_ERRORS 15

//
// Custom Thermistor 1000 parameters
//
#if TEMP_SENSOR_0 == 1000
  #define HOTEND0_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND0_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND0_BETA                 3950    // Beta value
#endif

#if TEMP_SENSOR_1 == 1000
  #define HOTEND1_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND1_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND1_BETA                 3950    // Beta value
#endif

#if TEMP_SENSOR_2 == 1000
[... 3433 lines cut ...]
#if ENABLED(PRINTCOUNTER)
  #define SERVICE_WARNING_BUZZES  3
  // Activate up to 3 service interval watchdogs
  //#define SERVICE_NAME_1      "Service S"
  //#define SERVICE_INTERVAL_1  100 // print hours
  //#define SERVICE_NAME_2      "Service L"
  //#define SERVICE_INTERVAL_2  200 // print hours
  //#define SERVICE_NAME_3      "Service 3"
  //#define SERVICE_INTERVAL_3    1 // print hours
#endif

// @section develop

//
// M100 Free Memory Watcher to debug memory usage
//
//#define M100_FREE_MEMORY_WATCHER

//
// M43 - display pin status, toggle pins, watch pins, watch endstops & toggle LED, test servo probe
//
//#define PINS_DEBUGGING

// Enable Marlin dev mode which adds some special commands
//#define MARLIN_DEV_MODE

Benderrodrigu3z, 2020-09-15:
ahh its a Safari Issue, here are the files
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5228133/Configs.zip)

Benderrodrigu3z, 2020-09-15:
You might notice I'm using the default code, EVEN the default code comes across as Japanese

GMagician, 2020-09-15:
Are you sure you are flashing correct compiled code? Once you set "language" it should change it

Benderrodrigu3z, 2020-09-15:
Im using the code found in the Creality Ender 5 Plus package, with Arduino, I click upload, note I tried changing the language to spanish and it had no effect on the panel

Benderrodrigu3z, 2020-09-15:
all i have is two screens, a screen that has stop in japanese and another tab, if I click the tab I see temps and z offsets in Japanese, also the back button doesnt work, only way out is to reboot

Benderrodrigu3z, 2020-09-15:
see video for what I have on the screen
[IMG_8975.mov.zip](https://github.com/MarlinFirmware/Marlin/files/5228172/IMG_8975.mov.zip)

GMagician, 2020-09-15:
mmm not sure but I think your lcd has a own firmware and language is not related to the one you program into Marlin.

Benderrodrigu3z, 2020-09-15:
crap, how do i fix that? new LCD? what kind offers a touch screen?

Benderrodrigu3z, 2020-09-15:
what is the default screen size on the ender 5 plus?

GMagician, 2020-09-15:
If what I suppose is correct, you may have 2 options:
1) upload a new firmware into lcd
2) change language from lcd itself (if there is a menu option and you are lucky to "translate" japanese/chinese)

How your lcd is connected to MB? Maybe a serial/i2c/spi bus? Then my guess is correct

Benderrodrigu3z, 2020-09-15:
it looks like a serial connection but i can change it to usb, how would I program it?
[image]
[image]

ellensp, 2020-09-16:
This is smart screen, controlled by Serial only. 
Display language is not controlled by marlin.

The firmware on the smart screen already contains English and  Chinese 
Click on the right hand cog looking item, settings
[image]

Next click the Bottom middle is a Icon (has a Chinese character with a right angle arrows and a E character) 
[image]

closed 2020-09-16 by ellensp