MarlinFirmware/Marlin #19405: [BUG] Uncomplete text on cr10s screen

Liger0, 2020-09-15:
### Bug Description

The screen on the cr10s lcs is not complete with 2 extruders, the extruder number is skipped.
Plus, the unload/unload options only change the first value, not the second.

### My Configurations

Bugfix as of 13 september

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5228232/Marlin.zip)

### Steps to Reproduce

1. Configurations-> Advanced->Filament

**Expected behavior:** Show the extruder number and change the values of the second filament if using the generic option

**Actual behavior:** The strings are the same for both the filaments, and the change of the generic ones only changes the first one

#### Additional Information

[image]

ellensp, 2020-09-16:
Confirmed on current bugfix.

ellensp, 2020-09-16:
Changing font is not an option.
scrolling the menu is a horrible idea.

In lanugaue_en.h is the following,  These are displayed in  Advanced->Filament and  Filament

  PROGMEM Language_Str MSG_FILAMENTCHANGE_E                = _UxGT("Change Filament *");
  PROGMEM Language_Str MSG_FILAMENTLOAD_E                  = _UxGT("Load Filament *");
  PROGMEM Language_Str MSG_FILAMENTUNLOAD_E                = _UxGT("Unload Filament *");

There is no need to repeat the word filament in the filament menu, I recommend 

PROGMEM Language_Str MSG_FILAMENTCHANGE_E                = _UxGT("Change *");
PROGMEM Language_Str MSG_FILAMENTLOAD_E                  = _UxGT("Load *");
PROGMEM Language_Str MSG_FILAMENTUNLOAD_E                = _UxGT("Unload *");

Liger0, 2020-09-16:
Is it normal that the generic option only changes the first value?

rhapsodyv, 2020-12-07:
The fix was merged! Thanks

closed 2020-12-07 by rhapsodyv